Zolko-123/FreeCAD_Assembly4 #218: Batch/List of datum imports

abetis, 2021-08-25:
It will help a lot if it will be possible to select multiple LCSs in the tree and by pressing import datum button the command will import them all with the default names.

Maybe a warning dialog to ask if you're sure default names will be used should be shown with OK/Cancel buttons.

I have parts with tons of holes, each have the HoleAxis/LCSs assigned and have to go one-by-one to import them :(

abetis, 2021-09-24:
@Zolko-123 
I see that datum import have a logic for 1 selected item in the tree (the one I generally use) and also for 2 selected object, that actually doesn't work since it's calling a non-existing Asm4.messageBox

Is it your code?
Don't really understand what its trying to do.
Should it still be there if it doesn't work?

Zolko-123, 2021-09-24:
oooops, its Asm4.warningBox ! Corrected, thanks (will be in the next release)

abetis, 2021-09-24:
Can you explain what does the code do when 2 datums are selected?
don't understand what you mean by superimpose there..

abetis, 2021-09-24:
Please see #245

closed 2021-10-04 by Zolko-123