Zolko-123/FreeCAD_Assembly4 #222: Fastener Placement Exception

kantlivelong, 2021-08-28:
Attempting to edit the placement of a fastener using ASM4 0.10.1 results in an exception. The previous version where this works that I tested was ASM4 0.9.17.

```
13:49:44  Running the Python command 'Asm4_placeFastener' failed:
Traceback (most recent call last):
  File "/home/shawn/.FreeCAD/Mod/Assembly4/FastenersLib.py", line 168, in Activated
    asmType = selection.AssemblyType

'FeaturePython' object has no attribute 'AssemblyType'
```

OS: Linux Mint 20.2 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 0d9536ed3e8c7f40197b5606e1b7873625e1d6fe
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

Assembly4 Version: 0.10.1

rmTizi, 2021-08-28:
I can confirm the issue happens on my setup as well

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

Assembly4 Version: 0.10.1

Zolko-123, 2021-08-29:
ooops, yes, sorry, I'll correct it

Zolko-123, 2021-08-29:
I pushed a new version 0.10.2 that should fix this bug: can you please update and test ?

kantlivelong, 2021-08-29:
I was able to insert a fastener however moving it doesn't appear to work.

This time the console simply shows
`08:08:39  This is not an error message you are supposed to see, something went wrong`
when selecting the fastener and clicking **Move/Attach part in assembly**

Also not sure if related but inserting parts no longer updates the preview when selecting LCS's, rotating, or translating.

Zolko-123, 2021-08-29:
That is the wrong command: you have to use [place Fastener](https://raw.githubusercontent.com/Zolko-123/FreeCAD_Assembly4/85ec934a5d2b12d45b92145395d19d3f17d16781/Resources/icons/Asm4_mvFastener.svg) for that

kantlivelong, 2021-08-29:
Oh right my mistake, in that case the **Edit Placement of a Fastener** is greyed out on the fastener.

[image]

Apologies if I'm doing something wrong, I've only been using ASM4 for a few days. :)

Zolko-123, 2021-08-29:
That's strange because it works on my computer. Can you close and restart FreeCAD, and test with a new fastener ? Are you sure you updated to the latest version ? (0.10.2)

kantlivelong, 2021-08-29:
[image]

```
Assembly 4 Workbench for FreeCAD-v0.19
0.10.2, 2021-08-29
```

Fasteners version indicates:
```
V0.3.34  14 Apr 2021:  Fix issue #93
```

Zolko-123, 2021-08-29:
yes of course, my mistake. There are other similar mistakes, I'll push an update ASAP

rmTizi, 2021-08-29:
FYI, 0.10.2 fixed the issue for my environment

Zolko-123, 2021-08-29:
pushed an update, can you please test ?

kantlivelong, 2021-08-29:
Yep I can move/attach fastener now :+1: 

Should I open a separate issue for the preview update when moving parts? Preview works fine for fasteners.

closed 2021-08-30 by Zolko-123