Zolko-123/FreeCAD_Assembly4 #234: Making Explosions Easier

xoviat, 2021-09-10:
Sometimes explosions are needed for assembly drawings. At first, I thought that this wouldn't be possible in FreeCAD, but it is with some effort:

[image]

If an assembly is created correctly, it's not to difficult to move the parts part, and then render that onto a drawing. The trickier bit is creating the dotted lines. For those, I use leader lines with the dotted style (without heads), but I need to manually edit the table to place them at the right angle (using some trigonometry). It should be possible to compute these leader lines, and maybe add an 'exploded' assembly configuration. 

Also, is it possible to render a configuration in a drawing without that configuration being active?

Zolko-123, 2021-09-11:
A drawing will always update to the current (active) configuration, unless it's set to "Frozen"  (or something similar, don't remember now). If you want several drawings showing several configurations (like an user assembly manual), then you need to make all your assembly steps showing the corresponding configuration, set the step's drawing to "Frozen", and go to the next step. 

As for the dotted lines: since the configurations are based on changing the AttachmentOffset, but since the AttachmentOffset is also used in other circustances, the configuration cannot "know" which is the assembled and which the exploded position. Therefore, it will not be possible in the near future to implement them.

closed 2021-09-11 by Zolko-123

xoviat, 2021-09-22:
One solution to this problem is to create two configurations: a base configuration and an exploded configuration. The difference in AttachmentOffset can be used to create the lines. I am not minimizing the implementation difficulty, but that seems to be the best path forward. What do you think?