Zolko-123/FreeCAD_Assembly4 #243: Part Placement Units Assume Metric

spontarelliam, 2021-09-23:
When placing a linked part, the X, Y, and Z Translation units are in mm, regardless of whether Imperial units are active in FreeCAD. This differs from how most length inputs work. For example, the Pad operation in Part Design assumes units that match the Preference units, and allows for overriding those unites by typing "1mm" or 1in".

FreeCAD version 0.19-22039, Assembly4 version 0.10.4

Zolko-123, 2021-09-24:
Yes, I don't know how to do this. And frankly, I don't care, nobody serious should use imperial units in engineering.

closed 2021-09-24 by Zolko-123

xoviat, 2021-09-28:
@Zolko-123 I can take care of this, if you want.

xoviat, 2021-09-28:
> nobody serious should use imperial units in engineering.

Personally, I can say that I work on/have worked on 'serious' engineering projects at several US companies and we have used imperial units. I am not claiming that these projects are/have been 'leading edge' but I have been/am paid to work on them.

Zolko-123, 2021-09-29:
NASA is using metric SI units

crashkopf, 2022-03-13:
> Yes, I don't know how to do this. And frankly, I don't care, nobody serious should use imperial units in engineering.

It's difficult to escape legacy unit systems.  Ever specified equipment that operates with vacuum pressure?  There's kPa, bar, atm, mmHG, mmWC, inWC, mmWC, Torr, and the scale can be either absolute or relative to atmospheric pressure.  You'd think all equipment manufacturers would use kPa, being an SI unit, but no.  When you're comparing equipment from different manufacturers you have to convert to common units.  Having to maintain these conversions by hand is a common source of errors.  NASA has experienced this, see [https://spacemath.gsfc.nasa.gov/weekly/6Page53.pdf](https://spacemath.gsfc.nasa.gov/weekly/6Page53.pdf).

What I'm getting at is that automated unit conversion can save a lot of headaches.  We can't always choose the units equipment is specified with, but we can choose how we work with them.

PR #250 (2021-09-29): add unit spin box
fixes #243
commit 173f31c2 (2021-09-29): add unit spin box
fixes #243