Zolko-123/FreeCAD_Assembly4 #254: Translation

FarmingSoul, 2021-10-01:
How make a update of \Resources\translations\asm4.ts for continue to translate the new dialog ??
Thankx all

Zolko-123, 2021-10-18:
I don't know

Zolko-123, 2021-10-18:
translation was added by @Thowil :

https://github.com/Zolko-123/FreeCAD_Assembly4/pull/214/commits/5005f483fa6e103d4f47fc98d2bcce1503b89b8f

I don't know how it works

hasecilu, 2024-02-06:
Hello, I can implement the translation for the WB, could you reopen the ticket?
I have a minimal implementation working.

Zolko-123, 2024-02-06:
done

Zolko-123, 2024-02-09:
hello @hasecilu , I've merged your translation code and it works indeed, thank-you for that :

> Inicializando en entorno de trabajo Assembly4 (0.50.8-dev) ............................. hecho.

I've modified the code to better suit my personal tastes (the function is called Qtranslate as before), so please update before making further changes. But I have 2 issues:

1) on my system, `pylupdate6` doesn't exist, so I can't update the language pack

2) more profound: I don't think that changing the menu entries is really that helpful, translating the explanatory tooltip would be much better for the user:

[image]

Changing the menu entry brings the disadvantage that the user cannot make an Internet search easily, and the default English tooltip is not translated. What I'd suggest is the opposite: leave the menu entry the default and provide translations for the tooltips. Is this doable ?

hasecilu, 2024-02-09:
> I've modified the code to better suit my personal tastes (the function is called Qtranslate as before), so please update before

As I mentioned on the PR review is not possible to use Qtranslate() because pylupdate is looking for instances of `translate(context, string)`, I tried with pyside2-lupdate, pylupdate5 and pylupdate6 and none of them catch Qtranslate().

> 1. on my system, `pylupdate6` doesn't exist, so I can't update the language pack

You can use `pylupdate5`, I used 6 because I found a problem with version 5 that doesn't pick strings: https://github.com/pyqt/python-qt5/issues/54
But if here we're not formatting the code probably will not be any problem.

> 2\. I don't think that changing the menu entries is really that helpful, translating the explanatory tooltip would be much better for the user:

I'm planning to fully translate the WB, but probably next week. But before that I want to know if you are planning on formatting the code or some kind of cleanup.

> Changing the menu entry brings the disadvantage that the user cannot make an Internet search easily, and the default English tooltip is not translated. What I'd suggest is the opposite: leave the menu entry the default and provide translations for the tooltips. Is this doable ?

Yes, both can be translated.

As you may know FreeCAD have a Help module where using the `What's this?` Command you can clic on a command on the toolbar and the wiki page for it will open. In this regard I think the best option is to create the Wiki Pages, It doesn't need to be super extensive articles.

Having said that, the way it works is: it opens `https://wiki.freecad.org/{$Command_name}`.

In the wiki the convention to name de pages is: `WorkbenchName_CommandName`, so in order to make this work, ideally your commands should be named the same.

For instance to create an InvoluteGear with FCGear WB you use `CreateInvoluteGear` command that opens https://wiki.freecad.org/CreateInvoluteGear/ when the documentation of that command is in https://wiki.freecad.org/FCGear_InvoluteGear article.

So apart from the repo itself, I think the best place to look for information about a command should be the FreeCAD Wiki, of course it would need to be worked on.

Zolko-123, 2024-02-11:
- OK, let's revert to `translate()`

- for `pylupdate6`, it would be important to check for the version. Also, it is really necessary to run it at every FreeCAD startup ?

- I don't particularly plan code formatting or clean-up, but if you think it's important then please separate PRs.

- also, please make small atomic PRs, it's easier to review the changes

- at this point, please don't translate the menu and function names entries, only the tooltips

hasecilu, 2024-02-24:
> * Also, it is really necessary to run it at every FreeCAD startup ?

You mean this? Yes because FreeCAD look for translation for the current locale, it they exist it uses them.
```python
    def Initialize(self):
        from TranslateUtils import translate

        FreeCADGui.addLanguagePath(Asm4_trans)
        FreeCADGui.updateLocale()
```

> * also, please make small atomic PRs, it's easier to review the changes

I plan to finish all tooltips this weekend, would you prefer 1 commit for every command, every 3?

hasecilu, 2024-02-25:
Hi, I sent new PR translating tooltips of the commands.
The tooltips of properties can also be translated but I'm busy now, I'll check later.
Also I think it's a good idea to also translate the name of the commands.

closed 2024-03-07 by Zolko-123