Zolko-123/FreeCAD_Assembly4 #261: No module named 'placeLinkUI'

alfredopalhares, 2021-10-10:
Hello,

I have just updated the assembly4 module. on FreeCAD 0,19 and get this message now ``No module named 'placeLinkUI'`
[image]

Here are my verions:
```
OS: Arch Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: makepkg
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.9.6
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
```

Assembly 4 commit:
```
$ git show HEAD
commit 1a68b6ed1175c25dfbc356c84eacb2accffd9391 (HEAD -> master, tag: v0.11.3)
Merge: 758a1ad 35b7510
Author: Zolko-123 <46596542+Zolko-123@users.noreply.github.com>
Date:   Sat Oct 9 22:54:38 2021 +0200

    Merge pull request #259 from Zolko-123/development
    
    v0.11.3
```
I did not change anything in the system, any help appreciated.

alfredopalhares, 2021-10-10:
Reverting back to v0.11.0 works:
´´´
 git checkout v0.11.0 
```

bdoering, 2021-10-10:
Duplicate of #260

Zolko-123, 2021-10-10:
yes, sorry, I forgot to merge the 2 files. Can you plaese update (to 0.11.3) and re-test ?