Zolko-123/FreeCAD_Assembly4 #270: assembly breaks upon small changes in sketch

hstarmans, 2021-10-31:
@Zolko-123  thanks for all the great work on Assembly4 . I used it with success, see below
[image]

My base plate is quite crucial for all the components attached to it.  If I change some of the dimensions in the base plate or add a hole, the naming of the edges changes.
This breaks all the links. Is there are fix for this? Can I ensure an edge in the sketch ends up with a certain name.
The challenge is also outlined on the [forum](https://forum.freecadweb.org/viewtopic.php?f=10&t=2267&sid=80c24a4af660f29d882a10007c9894b7&start=10).

Zolko-123, 2021-10-31:
Yes, there is a fix for this : in fact, one of the reasons-to-be for Assembly4 is to solve this exact issue. It's called "master sketch" and you can read about it in the forum and in the wiki (I don't know the links, sorry).

In short, here is what you do: 

1) instead of attaching the datum LCS to the geometry of the base-plate, you create a sketch in your Part or Body (a normal sketch, but Asm4 provides a convenient way to do it with your chosen custom name) in which you draw the function of your part : lines, circles, external shape .... In your case, you would draw in this sketch the places where you're going to put your parts. What is the best way to define this depends in the particular project: in can be holes, edges, center...

2) you design your part by taking this sketch as (external) reference: for example, when you make a hole in the plate where you're going to put some part, I suppose you make a sketch and then extrude a pocket with it through the plate that is already present. In this case, you import (into the sketch) as external reference the relevant edge/hole/whatever from your master sketch, and design your pocket with it. Thus, if you change the dimensions of the master sketch, the placement of that part will follow AND the hole will follow.

3) if your part is very complicated, you can make several master sketches, you're not limited to 1. But ensure that the attachment datums for your external parts are all referenced to these master sketches, and not the geometry, that you have also drawn with these master sketches as reference. 

4) Using the standard PartDesign WB  and Body (blue icon) produces a perfectly working component for Asm4, I recommend. If you have multi-solid stuff, and more complicated surface design, then you should use a Part (yellow icon) container and do your stuff in there. You can also use sub-assemblies if a component is itself made up of several items.

Does that make sense ?

Zolko-123, 2021-10-31:
BTW: nice design !

hstarmans, 2021-11-01:
@Zolko-123 thanks for the quick reply , I will use this than in my queries and assume I am able to fix it.. the coming days I am doing some different stuff but will provide an update once I fixed my model.

closed 2021-11-01 by hstarmans