Zolko-123/FreeCAD_Assembly4 #271: Name for the link is "Body"

v-h-giang, 2021-11-11:
After reinstall i got some strange behavior when insert a part to assembly model (ctrl + L) , the "Name for the link" is defaultly set to "Body" . It was "insert file name  +  '_Body' " before. When attaching a part in the assembly, in "Place linked Part" window, the "Attach to :"  list will not change with 3d view selection as before. I must select the attach part from the pop down list. 
Any something changed recently ?

Zolko-123, 2021-11-11:
For the filename, yes it was changed because it made very long names, and also for some cyrillik languages the filename characters where replaced with _ which was not helpful

for the selection, no, this is not normal, what version are you using ? What OS ?

v-h-giang, 2021-11-11:
I really like the old way that "name for the link" can automatic set to "filename_Body". I am using Assembly4 for designing and development machine with large amount of parts ( about hundreds) . Every parts have special name and id with its file name (in english).  Its really help if have some checkbox for old way naming , near "Name for the link" . 

About the selection , i am using newest stable version of Freecad (0.19 ) and Assembly4  , windows 10 with newest update. And that "Attach to :" list will change when click to LCS on the part only .  But almost the time , LCS stay in middle  of  Screw hole , on the surface that hard to click or view from outside of complex model .

Thanks for response !

Zolko-123, 2021-11-11:
I guess the automatic naming should be set in the user preferences. I'll look what I can come up with

leoheck, 2022-09-09:
I thought you were not into this automagical bs, lol.

[image]

https://github.com/Zolko-123/FreeCAD_Assembly4/issues/344

closed 2022-09-12 by Zolko-123