Zolko-123/FreeCAD_Assembly4 #272: BOM not working after updating assembly4 to version 0.11.4

markbil, 2021-11-22:
hi, thanks for the great work on assembly4! I am designing my kitchen cabinets in freecad, and since I updated assembly4 to version 0.11.4 from an older version through the addon-manager, the BOM feature (Menu => Assembly => Create Part List) does not seem to work any longer. in the older version a popup window would appear with a list of all my Part names (kitchen cabinet panels) and their dimensions, i.e. something similar to "Top Panel: 600 x 18 x 450, etc.". Now when I execute the same function I receive an empty popup as pasted below... Also the report view throws "10:28:40  3634.66 Selection.cpp(1649): Object not found". I changed nothing but updading assembly4 to the latest version...
Am I doing something wrong or is this a bug? Thanks for your help!
[image]

Zolko-123, 2021-11-22:
Thank-you for reporting, I'll have a look at it. The BOM functionality is mostly implemented by @FarmingSoul

@FarmingSoul : could you have a look at the issue at hand here ?

markbil, 2021-11-22:
@Zolko-123 great, thank you!

for what it's worth, I created the following macro which prints my part list and their dimensions as expected. Using BoundBox **some** of the dimensions are very slightly off (like 0,00000001), which I fixed with rounding. Not sure why and whether there is a better method to get the exact dimensions for each part...

```
doc = App.ActiveDocument
for object in (doc.findObjects("PartDesign::Body")):
    print (object.Label2, end=": ")	
    if hasattr(object, "Shape"):
        bbc = object.Shape.BoundBox
        zl = round(bbc.ZLength,3)
        yl = round(bbc.YLength,3)
        xl = round(bbc.XLength,3)
#       print (str(bbc.ZLength), str(bbc.YLength), str(bbc.XLength))
        print (zl, yl, xl)
```

FarmingSoul, 2021-12-06:
@Zolko-123 , @markbil 
Sorry this 3 weeks i work a lot.
sorry for the inconvenience and thank you for the return. 
In the new version the BOM fields are configurable. 
explanation :

Select a part
[image]

Click on Edit Part Information
[image]

Click on config
[image]

in this column you will find the default names. 
If you stay with the mouse on a field a tooltip opens to indicate the principle of this autofield.
[image]

here you can customize the name of the field
[image]

here you can decide if you want to use this autofield
[image]

this function is currently not present
[image]

you can add fields to be filled in manually in Edit Part Information
[image]

Your request does not have the area to be managed by the current autofield so I will integrate your macro and a new autofiled will be present in the next version of ASM4.
Here is the file I will modify.
[image]

here there is a typical explanation for creating your own autofield
[image]

Have nice day

FarmingSoul, 2021-12-06:
[image]

fix

you can use this file :

https://raw.githubusercontent.com/Zolko-123/FreeCAD_Assembly4/0e872345049d72f58f84fee6df03aa85640a614f/InfoKeys.py

and the next Update of ASM4 have this fields .

Nice Day

robertlinn0423, 2022-04-22:
I'm experiencing the same empty Bom panel when clicking the PartsList button as markbill shows on Nov. 22, 2021.  I'm running these versions:

- O/S: Linux Ubuntu 21.10
-  FreeCAD 0.19 revision 24366
- Assembly 4: 0.11.11
When I click the Ok button from the Parts List/BOM panel the console shows '6.82e-07 Selection.cpp(1649): Object not found'.
I've followed the tutorial from here: https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/Examples/ConfigBOM/README.md several times always with the same result 'Selection.cpp(1649): Object not found'

Any thoughts or help is appreciated.
Thanks
Bob

leoheck, 2022-09-29:
@robertlinn0423 the development branch has an improved BOM, I invite you to check if it works for you.

Zolko-123, 2022-09-30:
BOM has been updated recently, please see #341

closed 2022-09-30 by Zolko-123