Zolko-123/FreeCAD_Assembly4 #273: Parts Randomly rotate when changing Variable

defini-tiv, 2021-11-24:
Steps to Reproduce:

1. Create a part and add LCS
2. Make a float variable
3. Create sketch in assembly-container with a constraint referencing the variable
4. Make a LCS connected to the sketch
5. Attach object with to sketch using the tow LCS
6. Change value of variable in “Data”-tab of “Variables”-object

--> The Part will rotate (in my case 180° around global z)
This is not a consequence of the constraint referencing the variable changing! The sketch does what its supposed to do.
The flaw happens in the attachment between the tow LCS

Zolko-123, 2021-11-25:
This is due to FreeCAD's topological naming problem. There is nothing that the Asm4 workbench can do. You need to change the way you reference the LCS.

When rotating things, I attach the rotating LCS to some fix LCS or sketch, and put the rotation parameter into the LCS's AttachmentOffset.

defini-tiv, 2021-11-26:
I knew about the topological naming problem, but thought it would only appear when adding/deleting objects like faces?

I think what you describe is what i did. I have no rotation parameter in the sense of a property with adjustable value.
I have a fixed rotation offset that doesn't change. Or did you mean that?
My variable is for controlling a linear movement.

Zolko-123, 2022-02-03:
> The Part will rotate (in my case 180° around global z)

that's not random, that is a sure sign of either the topological naming or a wrong LCS attachment. Not Assembly4 related.

closed 2022-02-03 by Zolko-123