Zolko-123/FreeCAD_Assembly4 #279: Assembly4 Configurations Manager - Not working Properly ?

prokod, 2022-01-03:
**FreeCAD Version:**
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0

**Assembly 4 Version:**
0.11.4

**Assembly Workflow used:**
Single file Assembly containing both `Parts` and The `Assembly` with `links` to those `Parts`

**Expected:**
When creating a new `Configuration`, the respected `Configuration Spreadsheet` should contain:
1. All the Assembly `Model` linked `Parts`
2. `Linked Parts` respective `Visibility` state snapshot
3. Some default or otherwise non empty `Type` value under the respected spreadsheet column

**Observed:**
1. Only one linked Part is found to be entered in the `Configuration Spreadsheet`
2. I cannot seem to follow the exact inclusion pattern but at least it seems that Visibility state of that linked Part should be True
3. No default `-` value is under `Type` column which throws errors related to it in the report view.

**Work Around:**
1. Switching Off/On Visibility of a specific linked Part and creating afterwards a new transient Configuration
2. Copying the content of the transient Configuration Spreadsheet to the desired, all inclusive, target Configuration Spreadsheet as a new raw
3. Remove the transient Configuration
4. Manually adding default Type value `-` where missing

Zolko-123, 2022-02-15:
Hello, this should be fixed in the latest v0.11.9 version. Can you please test and report ?

prokod, 2022-02-15:
Hi @Zolko-123 I have tested and I can confirm that when I add new configuration All the available linked parts are present and both `Variable`, `Assembly Type` columns are populated as I would expect. Thanks!

prokod, 2022-02-15:
BTW, on the same topic, different aspect. Does Assembly4 supports Configuration nesting ?  
What I would expect is that I could add an sub "assembly4" Part (which contains multiple Configurations) to another "assembly4" Part and there I could influence which Configuration to use from the sub assembly either by a default or through yet again Configuration item within the higher level assembly.  
I did not find this mentioned in docs - if need be I can open feature request under Issues

prokod, 2022-02-15:
LGTM

closed 2022-02-15 by prokod