Zolko-123/FreeCAD_Assembly4 #281: Linked parts rotate together in tutorial

freelon, 2022-01-07:
Hi, I'm doing the example from [https://github.com/Zolko-123/FreeCAD_Examples/blob/master/Asm4_Tutorial1/README.md](https://github.com/Zolko-123/FreeCAD_Examples/blob/master/Asm4_Tutorial1/README.md). Weirdly enough, when I link the second bearing and rotate it to match, the first bearing gets rotated as well. If I randomly rotate one bearing around for instance the x axis through the `Edit placement of a part` menu entry, both inserted bearings are rotated. Since I'm new to assemblies (obviously, if I do the tutorial :P) I'm not sure if I just missed something or this is a real problem. Any help is appreciated.

I freshly installed Assebmly4 from the plugin manager. Freecad info:
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: makepkg
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.10.1
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

freelon, 2022-01-07:
So I played around with it further and here's what I found so far: it works well if I use the axis and bearing files from the git repository. I can place them and turn them independently. Only my self made ones seem to have the problem that all instances of them somehow rotate together.

* When I create the LCS' in the parts I get the following warning: `<built-in function SoSearchAction_getPath> returned a result with an exception set` popup, but after that I can attach the LCS to the edges.
* Then when I insert a link to the part the report shows the following error: 
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Part/AttachmentEditor/TaskAttachmentEditor.py", line 354, in addSelection
    prevref = LinkFromStr( self.refLines[i-1].text(), self.obj.Document )
  File "/usr/lib/freecad/Mod/Part/AttachmentEditor/TaskAttachmentEditor.py", line 93, in LinkFromStr
    raise ValueError(_translate('AttachmentEditor',"No object named {name}",None).format(name= pieces[0]))
<class 'ValueError'>: No object named Bear_Ringsdfasdf`
`

The same issue arises if I create multiple links to the axis, so it shouldn't have anything to do with the import..

Zolko-123, 2022-01-10:
yes, there is a bug somewhere, I'm looking into it, thank-you for the report

Zolko-123, 2022-01-10:
OK, found the bug: it was the Part::AttachExtensionPython introduced in the previous version that also creates an AttachmentOffset which conflicts with that of Asm4. Please update and test.

Unfortunately, the objects created with that version (0.11.5) cannot be automatically repaired, they must be redone. Sorry for that.

tfirchau, 2022-01-11:
I ran into the same problem yesterday, found this issue and subscribed to it.
Saw your reply and fix this morning. Did a manual install of the latest version of the add-on, re-did the tutorial, and it works!

Good Job! Thank You! 👍

Zolko-123, 2022-01-12:
@freelon : if it is fixed, can you please close the issue ?

freelon, 2022-01-12:
Sorry, I didn't have time to check it myself yet, I'll do today and then close it.

freelon, 2022-01-12:
It works for me, too. Thank you very much for the quick fix!

@Zolko-123 I didn't find anything, but do you take donations for your great work here?

closed 2022-01-12 by freelon

hpb-htw, 2023-02-08:
I have again the same exception at version 0.20.1

```
OS: Ubuntu 22.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 22f524c0c755003c12234fab1bad6a9f696b661a
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * Assembly4 0.12.5
``

Zolko-123, 2023-02-08:
@hpb-htw : can you please try by creating a new assembly and new parts and bodies ? It might be because you have used v0.11.5 sometime in the past.

hpb-htw, 2023-02-08:
Should I also remove `~/.FreeCAD` and `~/.local/share/FreeCAD`  and reinstall the plugin before I try to create them again?

hpb-htw, 2023-02-15:
Ok, I just redraw the sketch, reduce the complexity and the problem is gone. Of course I have removed `~/.FreeCAD` and `~/.local/share/FreeCAD`  before. I think the problem was in my sketch.

Zolko-123, 2023-02-16:
Thank-you for letting known.