Zolko-123/FreeCAD_Assembly4 #285: Assembly4 workbench breaks picking

maxlem, 2022-01-25:
Hello, 
Thanks for your plugin. I'm having an issue than I can't find reproduced anywhere on the www.

I install freecad 0.19 (e.g. sudo snap install freecad) or 0.19.3 (freecad-ppd)
I load my part (attached but it's the same behavior for a simple cube)
I can pick edges, faces, etc in the 3d view

I install Assembly4 from the add-on manager
I restart freecad
I can't pick anything anymore in the 3d view (egdes, faces, etc)

I noticed it trying to add an LCS to my part in order to refer to link it in an assembly.

I tried with master (build from souce), but then all Assembly4's  button are grayed all the time.

Thanks

maxlem, 2022-01-25:
hmm still investigating, I thinks it's something else unrelated, will update soon

Zolko-123, 2022-01-25:
This seems to be reported by other users as well. I think some bugs have been introduced in FreeCAD recently that does this

maxlem, 2022-01-25:
Ok I think I found the problem, it is the OpenInventor mouse navigtion style cause the picking to fail, I'll try to report it to freecad

closed 2022-01-25 by maxlem