Zolko-123/FreeCAD_Assembly4 #286: How to use the symmetry tool

maxlem, 2022-01-25:
Hello again,

I create a part and link it to an assembly
Then I create a symmmetric part from it
Then I add this _sym part to my assembly, positionning it somewhere else
now I see my _sym part at original location and the newly positionned one
When I try to hide the _sym part, the newly positionned part hides too...

Assembly Current version 0.11.6
OS: Ubuntu Core 20 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Unknown
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.0
Locale: English/United States (en_US)

Zolko-123, 2022-02-03:
The symmetry tool is very useful but not very reliable.

Zolko-123, 2022-02-03:
see #277

closed 2022-02-03 by Zolko-123