Zolko-123/FreeCAD_Assembly4 #291: Assembly 4 - Can't link to an external part more than once in an Assembly

buteomont, 2022-02-09:
If I create an assembly and insert a link to an external part once, all is well. However, if I insert a second instance of the same part, it works only if the second part is attached to a different LCS than the first one AND it doesn't have an offset. If I add an offset, the first part moves that same amount.  If I say, rotate one of the parts, **both** parts rotate. Worse, that part moves in _**ALL**_ assemblies that link to it, even those in other projects. Worse even still, if I play with the offsets to try to fix it, sometimes FC will crash and disappear.
Apparently, adding a link to an external part modifies the file of the external part, which kind of negates the whole reason to use external parts (part reuse).
I've found that I can work around this by creating a new assembly LCS for every part I add, and attaching the part to that LCS, and _never_ using an offset or rotation for any parts.   

I'm just learning FreeCAD and love it so far, but surely this is not correct operation.

Steps To Reproduce:

1. Create a new file in the Assembly 4 workbench and import a STEP file of something (e.g., a motor).
2. Drag the imported part into the Part container in the object tree.
3. Save and close the file, and then re-open it.
4. Create another new file and add an Assembly container. Save, close, reopen.
5. Right-click on the Assembly container, and choose Assembly -> Insert Link To A Part.
6. Select the part you created earlier and click Insert.
7. Attach it to the parent assembly, and choose the LCS of both items. Click OK.
8. Insert another link to the same part, in the manner described above, except this time add a 40mm offset in the X direction. Click OK.
9. Now notice that there is two parts, but both are congruent with one another and they are both at the 40mm X offset.

FreeCAD info:

OS: Ubuntu 20.04.3 LTS (GNOME-Flashback:GNOME/gnome-flashback-metacity)
Word size of FreeCAD: 64-bit
Version: 0.20.26554 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 817c14b)
Hash: 817c14b16af681b06aec7d8c79f744c7a88fa3c3
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)

Note: I submitted this [as a FreeCAD issue](https://tracker.freecadweb.org/view.php?id=4829) but was told to submit it here instead.

Zolko-123, 2022-02-09:
what Asm4 version do you use ?

buteomont, 2022-02-10:
> what Asm4 version do you use ?

I have the current version (0.11.8) installed.

Zolko-123, 2022-02-10:
version 0.11.5 had a bug, that actually corrupted the files it made, and which had that symptom. Did you create/use the faulty part with version 0.11.5 ? Does the problem persist with new parts ?

The only solution is to re-create the faulty part, sorry

Zolko-123, 2022-02-10:
I did the exact steps you describe, and it works as expected in Asm4. What you describe was the problem with version 0.11.5, an unfortunate bug, that also corrupted the files it created. Please test again with the current WB version and new files.