Zolko-123/FreeCAD_Assembly4 #302: No such command 'Part_EditAttachment'

s-em-i, 2022-03-22:
Hey @Zolko-123,

every time I am creating a new LCS I get hits error message:

```
15:23:09  Initializing Assembly4 workbench (v0.11.10) ............................ done.
15:23:40  Running the Python command 'Asm4_newLCS' failed:
Traceback (most recent call last):
  File "/Users/semi/Library/Application Support/FreeCAD/Mod/Assembly4/./newDatumCmd.py", line 159, in Activated
    Gui.runCommand('Part_EditAttachment')

No such command 'Part_EditAttachment'
```
The new LCS is created, but I have to manually click on **Edit datum**, because now window is popping up.
However, If I execute the command in the console, doesn't matter if with Assembly4 WB, Part or Part Design:
```
>>> Gui.runCommand('Part_EditAttachment')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
Base.FreeCADError: No such command 'Part_EditAttachment'
>>> Gui.runCommand("Part_EditAttachment")
Traceback (most recent call last):
  File "<input>", line 1, in <module>
Base.FreeCADError: No such command 'Part_EditAttachment'
```

Any Idea how I can fix it? Or what's causing this issue?

```
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.20.27936 (Git)
Build type: Release
Python 3.9.10, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * Manipulator
  * fasteners 0.3.38
  * Assembly4 0.11.10
```

s-em-i, 2022-03-22:
No it is not in there.
I checked that as well already.
Also strange is, that I don't even have the [Part EditAttachment](https://wiki.freecadweb.org/Part_EditAttachment) Icon in my [toolbar](https://github.com/FreeCAD/FreeCAD/blob/50f16686f1fd9813a099a380f2bdd3e243a0325b/src/Mod/Part/Gui/Workbench.cpp#L142).

[image]

I do have **Attachment Editor** in my context though.

How can i get the internal python command that start these context commands **Edit datum**
and **Attachment Editor** ?

Zolko-123, 2022-03-22:
I din't use MacOS, I'm on linux, so I'm not sure. May-be you have several FreCAD programs installed ? You can also remove your local FreeCAD config directory, may-be it was mixed-up ?

s-em-i, 2022-03-23:
I thought that too, but no I don't, just the one Version.
I've cleaned already and started all over, that is how I found out about this difference.
For now, I don't think this is really A4 related, I am sorry. If I find something out that says different,
I will reopen this issue. Thank you for checking in.

s-em-i, 2022-03-23:
Sorry @Zolko-123, I have to reopen the issue.
I think I have now more [evidence](https://forum.freecadweb.org/viewtopic.php?f=8&t=67387&p=582363#p582363) that points to an A4 related Issue.

s-em-i, 2022-03-25:
This is a FreeCAD Issue, not handling the passed file path correctly over the Apple Events.

closed 2022-03-25 by s-em-i