Zolko-123/FreeCAD_Assembly4 #303: Can't stop animation, `Animate Assembly` window blocked

hasecilu, 2022-04-02:
When you animate assemblies with a lot of parts (~70 in my case) and the `Sleep` time is near to 0 the cursor is always in loading mode and you can't press the `Stop` button, you can't interact with any of the elements of the window (except move, minimize, close) nor FreeCAD so there is no way to stop the animation (not even closing the window). The only option is to pkill FreeCAD.

With a bigger time (0.4s in my case) you can interact with UI elements and stop the animation.  The time needed for calculate every step depends on number of elements and CPU power.

* One solution is force a small `Sleep` time even if is 0, a small fraction o the time used to calculate positions.
* Another is don't lock the UI elements (or this is a OS thing?)
* If the `Animate Assembly` window is closed stop the animation, I think this is the easiest option

Zolko-123, 2022-04-04:
> With a bigger time (0.4s in my case) you can interact with UI elements and stop the animation. The time needed for calculate every step depends on number of elements and CPU power.

Good catch. I think that a default 0.1s (or 0.01s) of default sleep would not hurt, but would let the CPU free for enough time to interact with the GUI. Could you test with 0.1s and tell whether that makes a usable experience ?

 
>     * If the `Animate Assembly` window is closed stop the animation, I think this is the easiest option
 
this would indeed be nice but it's not that easy to implement

hasecilu, 2022-04-05:
> Good catch. I think that a default 0.1s (or 0.01s) of default sleep would not hurt, but would let the CPU free for enough time to interact with the GUI. Could you test with 0.1s and tell whether that makes a usable experience ?

My assembly have 4 animations: 
* one with 1 moving element: no problem with Sleep 0.0s
* another with 10 moving elements: no problem with Sleep 0.0s
* another with 29 moving elements: no problem with Sleep 0.0s
* another with 36 moving elements (Bed full): this one has problems with Sleep 0.0s to 1.2s, between 1.2 and 1.5s you need to make high speed clicking, 1.5 feels usable.

You can even check the assembly at [this repo]( https://gitlab.com/hasecilu/diy-pcb-cnc) btw, animations at Bed full, Vertical frame, XY and Z axis subassemblies.

> this would indeed be nice but it's not that easy to implement

=0

Zolko-123, 2022-04-11:
So, from your list, it looks like the default 0.0s sleep is good for most animations. Actually, what I do, is to always test the animation with "safe" parameters, making short animations, so when things go wrong I don't have to wait too long for the animation to stop. And only then do I try longer – full cycle – animations.

Could need a good tutorial to explain all this.

hasecilu, 2022-04-14:
That sounds good, I was using the `Pendulum` and `Loop` options so they will never stop.

Zolko-123, 2022-04-16:
so .... can we close this issue ?

hasecilu, 2022-04-16:
Yes

closed 2022-04-16 by hasecilu