Zolko-123/FreeCAD_Assembly4 #313: Assembly4 model gets corrupted when changing a parameter

Sil68, 2022-04-26:
Hi,

I'm working on a woodworking furniture project, and I've split up my model into several parts, one of them being a spreadsheet containing all the relevant dimensions and other information, another one being some kind of a channel.

When trying to re-use the channel assembly4 in another project, amending the relevant parameter in the parameter file, the channel model gets all corrupted, the various LCS getting changed in order and location, resulting in the parts getting placed wrongly (location, and angle).

The original model comprises the files parameter.FCstd and vchannel.FCstd, the amended on parameter2.FCstd and vchannel2.FCstd. The difference consisting in changing "CarcassDepth" (Carcass01Depth) from "400" to "500".

The former one being perfectly fine, the latter one having gotten damaged after re-computing the assembly, and opening the master sketch again.

Original assembly
[image]
[original_assembly.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/8566218/original_assembly.zip)

Amended assembly
[image]
[amended_ assembly.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/8566219/amended_.assembly.zip)

My environment comprises of
[code]
Macmini Apple M1
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.28671 (Git)
Build type: Release
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * CommandPanel
  * fcgear 1.0.0
  * fasteners 0.3.40
  * Autoload
  * Assembly4 0.11.10
[/code]

Btw. is there any better way of creating such an object than the approach I have taken?

al-right, 2022-06-11:
Your model is "hit" by the famous TNP (topological naming problem).
https://wiki.freecadweb.org/Topological_naming_problem :
> In Workbench [PartDesign](https://wiki.freecadweb.org/PartDesign_Workbench).svg PartDesign, if a feature is supported on a face (or edge or vertex), the feature may break if the underlying solid changes size or orientation, as the original face (or edge or vertex) may be internally renamed.

If you try to hover over e.g. the edges of your "sketch_master" sketch and remember the edge numbers whereafter you change the "CarcassDepth" parameter then you will notice that the e.g. edge numbers will change. The same usually happens with the vertex numbers. And that happens to be the case in your sketch as well.

The TNP likely will get fixed in FreeCAD 0.21 when the fixes of RealThunder et al. get into FreeCAD.

What you can do until then is to use more sketches to avoid the problem.
One take is to have a sketch for all the static stuff (those not affected by parameters).
And for every dynamic parameter affecting the original sketch create a new sketch so that parameter only affects one part so that that and another part one change position etc.
Now that might be one troublesome extra work.
Since I do not know your model and the details well you might want to ponder a bit about what the best strategy is.
However in the FreeCAD forum other people have also posted about ways to overcome the TNP for now.
I myself have had to overcome it one way or the other and really look forward to have it fixed, but so far there has always been a way to circumvent the problem. :)
In other circumstances you might want to use a sub-assembly to divide-and-conquer the problem into smaller tasks.

In your case I would start creating some extra sketches to separate side left/right from the wedges and experiment to see how it goes. If things start to get too complicated (e.g. too many sketches) create one or more sub-assemblies that are parameterized using "Create a variant Part" The yellow green icon. In that way you may make one sub-part longer/shorter and put it together with other sub-parts thereafter in a common/final assembly. You might want to take a look here: https://forum.freecadweb.org/viewtopic.php?t=62767

I hope it was understandable?

Cheers :)

Sil68, 2022-06-13:
Hi,

Thanks for you comprehensive reply & suggestions! Much appreciated!

I really would like to use realthunder’s build on my Mac M1, but unfortunately it’s crashing quite oftentimes. :(

Cheers!

closed 2022-06-20 by Zolko-123