Zolko-123/FreeCAD_Assembly4 #315: Crash on making or opening documents using variant link, from assertation failure in ExpressionImporter()

dicethrow, 2022-05-26:
Hello all,

I was following [this example project](https://forum.freecadweb.org/viewtopic.php?f=9&t=62767) and downloaded and opened the project files. All opened fine except for `MainAssembly.FCStd`, which after a second or two crashed.

Steps to repeat:
1. Build freecad from source, using the `FreeCAD-0-19` tag (commit reference 0a49fd05a4)
2. Start freecad from the terminal
3. When the freecad GUI has opened, open the file `MainAssembly.FCStd`. OR instead of opening the file, open several parts, then make a new empty assembly4 assembly and click the `create a variant link` button and select a part to link to.
After a second or two, the crash will then occur. Here is what is shown in my terminal:

```
$ FreeCAD
FreeCAD 0.19, Libs: 0.19R24267 +149 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.11.12) loaded
connect failed: No such file or directory
Exception opening file: /home/tmp/FreeCAD_Doc_7b3352ac-298f-434b-b6f8-d8543f4ea788_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_7b3352ac-298f-434b-b6f8-d8543f4ea788_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_2c7ee08f-d7b5-4a98-9cf5-8dcbb3dd2aab_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_2c7ee08f-d7b5-4a98-9cf5-8dcbb3dd2aab_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_de6cd91d-6a4d-4046-b8c9-c162d8ec021f_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_de6cd91d-6a4d-4046-b8c9-c162d8ec021f_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_5f5b96ee-28a8-4318-bbd2-ab76ada3cba3_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_5f5b96ee-28a8-4318-bbd2-ab76ada3cba3_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_e8ec0654-805f-4638-a292-e9575e396f5e_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_e8ec0654-805f-4638-a292-e9575e396f5e_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_e66e5e8f-fcc1-4826-bd92-6500a0cbd26f_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_e66e5e8f-fcc1-4826-bd92-6500a0cbd26f_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_5448cdf6-1734-455c-a34c-7c749e7c797d_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_5448cdf6-1734-455c-a34c-7c749e7c797d_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_9bc8541e-9ace-483e-8465-d9e81889d397_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_9bc8541e-9ace-483e-8465-d9e81889d397_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_a685fa17-f784-47df-a959-72e15e84fe55_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_a685fa17-f784-47df-a959-72e15e84fe55_da39a3_62388' does not exist!]
Exception opening file: /home/tmp/FreeCAD_Doc_521a1388-2ba5-43b8-98f9-ac9a300fcff0_da39a3_62388 [File '/home/tmp/FreeCAD_Doc_521a1388-2ba5-43b8-98f9-ac9a300fcff0_da39a3_62388' does not exist!]
1.397e-06 <App> Document.cpp(2854): Aggregate#PartB.LinkedObject: Time stamp changed on link PartB#PartB
0.000104974 <App> Document.cpp(2854): Aggregate#PartB.ExpressionEngine: Time stamp changed on link PartB#LCS_0
0.000141152 <App> Document.cpp(2772): Reload partial document Aggregate
PartDesign::CoordinateSystem / LCS_1_TEST: Links go out of the allowed scope
PartDesign::CoordinateSystem / LCS_1_TEST: Links go out of the allowed scope
FreeCAD: /home/x/Documents/git_repos/forks/mechanics/tools/freecad-source/src/App/Expression.cpp:3109: App::ExpressionParser::ExpressionImporter::ExpressionImporter(Base::XMLReader&): Assertion `!_Reader' failed.
Aborted (core dumped)
```

I also tried to use the development version of Assembly 4 by doing this (is this a valid way to do that?)
1. opening a terminal in `~/.FreeCAD/Mod/Assembly4/`
2. running `git checkout development`
3. repeat the above steps to start freecad
but no change in behaviour.

I hope I'm doing something wrong and that this is a straightforward fix. Any ideas?

Here's the output of pressing the About button in freecad:
```OS: Ubuntu 21.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +149 (Git)
Build type: Unknown
Branch: FreeCAD-0-19
Hash: 0a49fd05a46447aba70b6f18b5bb8210175dd471
Python version: 3.9.7
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/Australia (en_AU)```

Zolko-123, 2022-05-29:
> Build freecad from source

that's very courageous from you, but could you please verify this problem (the crash) on a release version ? (the error when opening variant link files is "normal" (as in "documented") but it still works)

dicethrow, 2022-06-01:
I'm surprised but that did it! any idea why?

Steps:
1. Install freecad, here's my terminal output
```bash
x@E595 $ sudo apt-get install freecad
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gnome-panel-data libcgi-fast-perl libcgi-pm-perl libconfig-general-perl libdbi-perl libdbi1
  libemail-date-format-perl libfcgi-bin libfcgi-perl libfcgi0ldbl libfwupdplugin1 libgnome-panel0
  libhttp-server-simple-perl libmime-lite-perl libmime-types-perl librrd8 librrds-perl libxml-simple-perl
  rrdtool
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  calculix-ccx freecad-common freecad-python3 libfreecad-python3-0.19 libspooles2.2
  python3-pyside2.qtuitools python3-pyside2.qtxml
Suggested packages:
  povray
The following NEW packages will be installed:
  calculix-ccx freecad freecad-common freecad-python3 libfreecad-python3-0.19 libspooles2.2
  python3-pyside2.qtuitools python3-pyside2.qtxml
0 to upgrade, 8 to newly install, 0 to remove and 4 not to upgrade.
Need to get 49.6 MB of archives.
After this operation, 211 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://au.archive.ubuntu.com/ubuntu impish/universe amd64 libspooles2.2 amd64 2.2-14 [466 kB]
Get:2 http://au.archive.ubuntu.com/ubuntu impish/universe amd64 calculix-ccx amd64 2.17-3 [1,825 kB]
Get:3 http://au.archive.ubuntu.com/ubuntu impish/universe amd64 freecad-common all 0.19.1+dfsg1-2 [17.2 MB]
Get:4 http://au.archive.ubuntu.com/ubuntu impish/universe amd64 python3-pyside2.qtxml amd64 5.15.2-1 [114 kB]
Get:5 http://au.archive.ubuntu.com/ubuntu impish/universe amd64 python3-pyside2.qtuitools amd64 5.15.2-1 [141 kB]
Get:6 http://au.archive.ubuntu.com/ubuntu impish/universe amd64 libfreecad-python3-0.19 amd64 0.19.1+dfsg1-2 [29.9 MB]
76% [6 libfreecad-python3-0.19 19.8 MB/29.9 MB 66%]                                                                                                      2,087 kB/Get:7 http://au.archive.ubuntu.com/ubuntu impish/universe amd64 freecad-python3 amd64 0.19.1+dfsg1-2 [19.5 kB]                                                   
Get:8 http://au.archive.ubuntu.com/ubuntu impish/universe amd64 freecad all 0.19.1+dfsg1-2 [6,264 B]                                                             
Fetched 49.6 MB in 28s (1,800 kB/s)                                                                                                                              
Selecting previously unselected package libspooles2.2:amd64.
(Reading database ... 307169 files and directories currently installed.)
Preparing to unpack .../0-libspooles2.2_2.2-14_amd64.deb ...
Unpacking libspooles2.2:amd64 (2.2-14) ...
Selecting previously unselected package calculix-ccx.
Preparing to unpack .../1-calculix-ccx_2.17-3_amd64.deb ...
Unpacking calculix-ccx (2.17-3) ...
Selecting previously unselected package freecad-common.
Preparing to unpack .../2-freecad-common_0.19.1+dfsg1-2_all.deb ...
Unpacking freecad-common (0.19.1+dfsg1-2) ...
Selecting previously unselected package python3-pyside2.qtxml.
Preparing to unpack .../3-python3-pyside2.qtxml_5.15.2-1_amd64.deb ...
Unpacking python3-pyside2.qtxml (5.15.2-1) ...
Selecting previously unselected package python3-pyside2.qtuitools.
Preparing to unpack .../4-python3-pyside2.qtuitools_5.15.2-1_amd64.deb ...
Unpacking python3-pyside2.qtuitools (5.15.2-1) ...
Selecting previously unselected package libfreecad-python3-0.19.
Preparing to unpack .../5-libfreecad-python3-0.19_0.19.1+dfsg1-2_amd64.deb ...
Unpacking libfreecad-python3-0.19 (0.19.1+dfsg1-2) ...
Selecting previously unselected package freecad-python3.
Preparing to unpack .../6-freecad-python3_0.19.1+dfsg1-2_amd64.deb ...
Unpacking freecad-python3 (0.19.1+dfsg1-2) ...
Selecting previously unselected package freecad.
Preparing to unpack .../7-freecad_0.19.1+dfsg1-2_all.deb ...
Unpacking freecad (0.19.1+dfsg1-2) ...
Setting up libfreecad-python3-0.19 (0.19.1+dfsg1-2) ...
Setting up libspooles2.2:amd64 (2.2-14) ...
Setting up freecad-common (0.19.1+dfsg1-2) ...
Setting up python3-pyside2.qtxml (5.15.2-1) ...
Setting up calculix-ccx (2.17-3) ...
Setting up python3-pyside2.qtuitools (5.15.2-1) ...
Setting up freecad-python3 (0.19.1+dfsg1-2) ...
update-alternatives: using /usr/lib/freecad/bin/freecad-python3 to provide /usr/bin/freecad (freecad) in auto mode
Setting up freecad (0.19.1+dfsg1-2) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for gnome-menus (3.36.0-1ubuntu1) ...
Processing triggers for libc-bin (2.34-0ubuntu3.2) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for shared-mime-info (2.1-1) ...
Processing triggers for mailcap (3.69ubuntu1) ...
Processing triggers for desktop-file-utils (0.26-1ubuntu2) ...

```
2. Repeat the 'method 2 to trigger the crash' from above, here is my terminal output
```bash
x@E595 $ freecad
FreeCAD 0.19, Libs: 0.19R
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.11.12) loaded
connect failed: No such file or directory
Initializing Assembly4 workbench (v0.11.12) ............................ done.
Initialising ...
Attaching VariantLink ...
x@E595 $ 

```
it didn't crash! amazing. Thanks heaps.

dicethrow, 2022-06-01:
I'd say that this is still an issue until we know why the crash happened, but I don't think it's a priority, so I'll close this for now.

dicethrow, 2022-06-01:
Interestingly, on the topic of differences between self-compiled and release-version of freecad: using the release version, the graphics seems snappier, and using the animation/variables slider is definitely less laggy / has higher fps rate. Not sure why.

dicethrow, 2022-06-01:
I wasn't able to work out what 0.20 actually is, I couldn't find a git tag or any reference to 0.20 in git. Does it just mean the current state of the master branch? 
But I did try to stick to the 0.19 git tag in an attempt to get an identical build to what the release version would be... but I'm not sure why the behavior is different, I'll need to spend more time understanding on this later. For context, I would like to contribute to the freecad code/plugins at some point.