Zolko-123/FreeCAD_Assembly4 #326: How do you version parts and assemblies?

leoheck, 2022-08-15:
Hi, I am looking for a way to maintain parts, assemblies, and versions...

Does this Assembly4 workbench has something that gives support to something like that? Or do you know if is there any workbench or way of doing that?

My idea is basically having a file for every part and part version, and a new version of the assembly file for every new version of the product.

Suppose I am creating this `Assembly-0.1` of a product. 

Then, the product uses some other parts, lets say, `A-0.1`, `B-0.1`, `C-0.1`, `D-0.1`

Then we have this first assembly 

```
- Assembly-0.1
  - A-0.1
  - B-0.1
  - C-0.1
  - D-0.1
```

Next time, an issue on A and B requires a change...

Then I would copy files `A-0.1` and, `B-0.1`, to `A-0.2`, and `B-0.2` and then apply the new changes.
After that, I would copy the `Assembly-0.1` to `Assembly-0.2` and then change the versions (filenames/paths) of parts that have a new version. 

In this way, I may have a version of each file. And also a working assembly of each time part is modified.

And then, maybe this way it would be good and easy and simple to store and keep track of these things when working with others. 

So, the question is more like, how easy is it to replace the file reference of each part, when they change the name (or the version number, actually)? Maybe it is quite easy as changing the path in the object's properties. But I am also checking if there is something like it does this automatically. Maybe looking for all the files with higher numbers that are in the same folder. Ah, maybe this can be done easily with a macro and then later could turn into a button.

leoheck, 2022-08-15:
After some time we would have these files and versions:

```
A ----: A1 ----> A2
B ----: B1 ----> B2 ----> B3
C ----: C1
Asm --: Asm1 --> Ams2 --> Asm3
```

Looking into another perspective each assembly would have these files.
```
Asm1: A1, B1, C1
Asm2: A2, B2, C1
Asm3: A2, B3, C1
```

Zolko-123, 2022-08-16:
What you're describing is a PLM/PDM system, and Assembly4 doesn't (yet ?) support anything of the like.

leoheck, 2022-08-16:
Alright. Anyways, @Zolko-123, thanks for teaching me those keywords.

closed 2022-08-16 by leoheck