Zolko-123/FreeCAD_Assembly4 #329: Multiple Animations / Explode Assembly

leoheck, 2022-08-17:
Is it possible to have multiple animations on Assembly4?

Or is it possible to create the exploded assembly with this workbench?

Maybe the exploded assembly could be something we could get out-of-the-box, where a sliding button would control the space between parts separating them from each other like the universe was expanding around them splitting the whole assembly apart.  =)

A similar thing is found on Grabcad Workbench

Before expanding
[image]

Expading the whole thing...
[image]

Zolko-123, 2022-08-18:
Exploded assemblies can be done with configurations: first, make a default configuration with all parts in their correct location. Then, move and/or hide parts as wished by adjusting the AttachmentOffset property of each part. You can then create a new configuration based on this state. As opposed to an exploded assembly, you can create as many states as you wish, showing/hiding parts, and/or offsetting parts from their assembled position. 

The goal of the configuration manager is to create user instructions for assemblies, which includes, but is not limited to, exploded assemblies

leoheck, 2022-08-18:
This looks interesting. Do you have a demo/example showing this workflow? I am using Assembly4 for 2 days only.

Zolko-123, 2022-08-24:
no, sorry

closed 2022-08-24 by Zolko-123

leoheck, 2022-08-27:
Indeed, configurations are awesome. Thanks for the instructions above.