Zolko-123/FreeCAD_Assembly4 #330: There is a place where LCS is not bein placed concentric to a hole.

leoheck, 2022-08-17:
Is it normal to fight a bit when creating an LCS? Sometimes it looks like the LCS does not move to the expected place. Of course, there are many attachment systems that I have no idea how they work yet. 

Then I am applying manual offsets and rotation to move my LCS to the expected place. Is this a good practice or this should be avoided?

I use to create assemblies by moving and positioning objects myself. I am experimenting with Assembly4 to see how better this design process can be.

For instance, I am trying to convert a manual assembly of a device that I made myself in the past. I got some parts in the right place, which is nice since I could learn how to use this workbench but one part has a hole for a board and a screw that no matter what I do, the LCS of a concentric circle is not being placed in the center of this hole. It is even not being placed close to the circle.

Since I don't master the LCS and I might have be doing something wrong or even my part might have issues that I am not aware of yet, I would like to check with you if this is something that happens eventually, Or if you think that it may be a bug related to Assembly4. In this case, I will share my design here as soon as it is finished. so you can give it a look too.

Zolko-123, 2022-08-18:
Converting a manually placed assembly to an attachment based assembly is probably not straightforward, because of core FreeCAD stuff (Placements of nested parts/containers for example). Also, Asm4 deals with linked objects, your assembly might contain proper parts, mixing such hybrid combinations could lead to unexpected results

Without the model it's difficult to say what could have gone wrong.

closed 2022-08-18 by Zolko-123

leoheck, 2022-08-18:
Nice, It looks like you haven't even read the post.

Zolko-123, 2022-08-18:
Or I have read it and it didn't make much sense. For example, what does "I got some parts ..." mean ?

leoheck, 2022-08-18:
Ah, then it is a different thing. You are asking a question politely, instead of being stupid.

I am not a native English speaker, unfortunately. So, sometimes I am going to say something that is hard to understand. 

I am using this Workbench for the first time since yesterday. I chose to reassemble one of my old designs, which was assembled manually. It took me some time to understand the LCS thing, because sometimes the coordinates do not move to where I want. And I am not sure if this is a Bug with Freecad, Bug with Assembly4, or just a bug on my parts. What I know is that Freecad did not complain about my parts, ever. But they were also made with previous versions of Freecad and I am also using Freecad beta (0.21). So, there are some variables to consider here. 

Now, with that being said. I could assemble some of my custom parts together. But one of them was giving problems to place those LCS, especially the concentric ones. 

Long story short. I could find later that this particular part did not have the Supports on 2 Sketches. Adding XY_Plane there fixed the issue. I am about to finish this assembly, I just have to understand how to place fasteners in a good way (the way you have explained in the other issue, thank you).

After that, since I am being able to use this Workbench, which is awesome by the way, I am going to share the design here with you. Just give me some minutes or a bit more to finish this.

leoheck, 2022-08-18:
Here is my design. I did not place fasteners yet. 
https://www.dropbox.com/s/nn2t6ohv2f057g2/peephole-v0.2.2_fixed_to_remove.FCStd?dl=0