Zolko-123/FreeCAD_Assembly4 #331: Interference check

leoheck, 2022-08-17:
Does Assembly4 have any way to check for interferences? That would be a great feature.

Zolko-123, 2022-08-18:
no, it doesn't, and I wouldn't know how to do this right now.

closed 2022-08-18 by Zolko-123

leoheck, 2022-08-18:
I see, this would be a good addition to this workbench.

leoheck, 2022-08-18:
Chrisb gave this idea, where we could do a boolean operation (maybe like boolean intersection), and check if the result is empty.

https://forum.freecadweb.org/viewtopic.php?f=3&t=71198

Now, a routine could iterate through the models inside "Model" assembly checking if they interfere with any other using this method. It does not need to be fast and perfect to start with, but it can be definitely improved with time. 

However, this interaction would be something that takes some time since, initially, all objects would have to be checked against every other unless there is something that it is possible to use to exclude some comparisons.

Actually, it is possible to use the part Volume, that I am seeing with the Part Information button, and the current coordinates of the object to exclude some comparisons. 

This would indeed be a really good feature.

leoheck, 2023-03-17:
Hi @Zolko-123 , I made something to give this power to Asm4.

Here, the Asm4 Model is the Swing.
The interference folder has the produced data.
And here is the output. 
It colors areas where objects interfere, so it is easy to spot where to fix.

[image]

Are you interested in having this embedded on Asm4 toolbar?