Zolko-123/FreeCAD_Assembly4 #332: Working with Fastners Workbench

leoheck, 2022-08-17:
With the fasteners workbench, it is possible to click at the hole then it is going to add the faster for me at that place.

Now, how should I work with the fasteners that are in the Assembly4 toolbar?
Should I make the Yellow body? Should I create LCS on every place I want to place a screw?
This is going to increase a lot the number of operations, especially if I have a lot of screws.

Assembly4 could have a faster way to add the screws as Fasterners Workbench does. 
Or maybe I am missing something.

Zolko-123, 2022-08-18:
You can use fasteners from the Fasteners WB directly if you prefer, they are in Asm4 only for convenience.

Unfortunately, the Fasteners WB doesn't recognize inserted objects/parts by links (App::Link) correctly, therefore you will not be able to attach a fastener to a hole in a child part of the assembly. 

The workflow to use fasteners inside the Asm4 WB is to select the hole edge you want to attach-to, create a "Hole Axis" feature (which is actually a datum line), and then use the "place part" button to attach the fastener to that hole axis feature. Yes, it's a little cumbersome, not as practical as what the Fastener WB does, but at least it can treat linked objects (which Fasteners WB can't)

Now, by playing with it to demonstrate, I realise that when selecting a "hole axis" feature, the fastener should be directly attached to it, instead of having to go through all the hoops. I'll try to improve this

leoheck, 2022-08-18:
Interesting, thank you. 

Is there a place where this workflow is shown for other users? 

The `README.md` has a bunch of info that is useless like Release Notes, that don't have to be there, for instance. 

Maybe a demo showing how this workflow works would be a good idea.

Ah, the Release notes can be put in a file itself and named `Changelog.md`, maybe. You can link this file in the `README.md` too.

Zolko-123, 2022-08-18:
Good ideas here. If you have the time, may-be could-you propose these modifications in a PR ?

leoheck, 2022-08-18:
Indeed. https://github.com/Zolko-123/FreeCAD_Assembly4/pull/333

leoheck, 2022-08-18:
> Now, by playing with it to demonstrate, I realise that when selecting a "hole axis" feature, the fastener should be directly attached to it, instead of having to go through all the hoops. I'll try to improve this

Nice. I was trying to replicate this workflow here but I am not sure if I have to put a screw inside the Yellow Body. I would like to take advantage of that so I won't have duplicated parts.

leoheck, 2022-08-18:
Talking about this workflow. 

Should the fastener be put inside the Yellow Body too? I am doing that. 

When I place those Hole Axis, they go inside the Blue Body, I could not keep them at the same level as the LCS. Is this expected? 

I could also create these "Hole Axis (datum lines)" but they don't appear in the `Place Linked Part` in the tasks view. Should they appear there?

Could you share a small demo?

This is a demo that a user shared with me in the Freecad Forum. But this is not the way you are describing here, I think.
However, you could use this easily the workflow described here.
https://forum.freecadweb.org/download/file.php?id=198783

leoheck, 2022-08-18:
This is my finished model. I am not using the above workflow. 
I would, but I could not understand it completely, yet.
https://www.dropbox.com/s/ai0cjb8zmgzbsfi/peephole-v0.2.2.FCStd?dl=0

Zolko-123, 2022-08-31:
Issues are for reporting problems, not for general questions or help requests. Please use the FreeCAD forum : 

https://forum.freecadweb.org/index.php

closed 2022-08-31 by Zolko-123

leoheck, 2022-08-31:
Sorry mate. I cannot use Freecad Forum. Have a good day.

leoheck, 2022-09-10:
Quick question.

Your workflow did not say how do you instantiate Fastners.
Should I put them inside the "New Part"?