Zolko-123/FreeCAD_Assembly4 #334: LCS placement improvement - Keep the Yellow Body selected after placing LCS.

leoheck, 2022-08-18:
I am working on placing LCS. It happens that I want to place more than 1 LCS in a single part. Which is normal, and may happen eventually. 

So, these are my steps, please, correct me if I am doing something not expected.

1. I click the Yellow Body (selecting it) 
2. Click the "New coordinate system"
3. Configure the attachment mode
4. Click OK/Cancel to finish the process

Then, the task menu closes and I don't have my Yellow body selected anymore. Which makes the process a bit hard. It should be still selected as I did before as starting this process. 

Now, The Blue Body inside is expanded (which is not what I would ever want) and the selection is not the Yellow Body anymore. So, keep adding LCS is tedious process, since I have to fold the Blue Body again since it has a huge list of features (this is optional, but I would prefer to keep the Body closed since I am not working with it) and then scroll up to the Yellow Body to add a new LCS.

So my question is. After placing an LCS, can the selection be restaured to where it was at the beginning of this process which is in the Yellow Body?

leoheck, 2022-08-18:
Ah, an alternative idea would be to allow placing the LCS in the Yellow body, having anything inside of that body selected. This would be also a flexible approach.

Zolko-123, 2022-08-24:
> Then, the task menu closes and I don't have my Yellow body selected anymore.

yes, because that task is _Part_EditAttachment_  and is an internal built-in task. Asm4 can't control the outcome of that task, and not the final state of its selection

> So my question is. After placing an LCS, can the selection be restaured to where it was at the beginning of this process which is in the Yellow Body?

unfortunately, no

closed 2022-08-24 by Zolko-123