Zolko-123/FreeCAD_Assembly4 #336: Asm4_NewPart with multiple Bodies

leoheck, 2022-08-23:
Hi Zolko, is it possible to have more than one Body inside the `Asm4_NewPart`?
I have a Part with 3 Bodies but I am getting the following message. I don't know what to do. Is this an error and has to be fixed? How should I proceed?

```
20:00:52  PartDesign::CoordinateSystem: Link(s) to object(s) 'Pocket023' go out of the allowed scope 'LCS_LIDAR_Connector'. Instead, the linked object(s) reside within 'Part_2 Part_2'.
```

Zolko-123, 2022-08-24:
> How should I proceed?
> ```
> 20:00:52  PartDesign::CoordinateSystem: Link(s) to object(s) 'Pocket023' go out of the allowed scope 'LCS_Connector'. Instead, the linked object(s) reside within 'Part_2 Part_2'.
> ```
Issues here are for listing and solving software bugs. For general usage help please address your questions to the FreeCAD forum : https://forum.freecadweb.org/viewforum.php?f=3

closed 2022-08-24 by Zolko-123

leoheck, 2022-08-25:
This [image] is definitely a question for here since the README does not have such info. And there is no info related to that, here on this repo.