Zolko-123/FreeCAD_Assembly4 #337: Select the configuration while Importing Part

leoheck, 2022-08-29:
Configurations are one hell of an important feature. Good job with this idea. Or maybe this is a design standard. But it is indeed awesome. Now, I could not select or find a way to import a specific configuration when using a model from another file. 

For instance. I have a design that has a stand for boards. This stand allows placing boards in 2 different ways. The design has already the boards mounted with 2 configurations. However, I could not select the configuration while instantiating the Assm4 model from another file. 

If this already exists excuse me but I could not find it.

This feature would definitely boost the Assembly 4 usability.

Zolko-123, 2022-08-30:
configurations are saved in a spreadsheet. Did you try to copy the configuration spreadsheet from 1 document into another ?

Zolko-123, 2022-08-30:
ah wait, now I understand. No, today it's not foreseen. You can set configurations in imported parts, but they will apply to all instances of that part

leoheck, 2022-08-30:
So, the only alternative is to create a second file then, since I cannot have 2 "Model" in the same document and since the configuration cannot be selected when importing the part. 

Do you have plans for such a feature?

Zolko-123, 2022-08-31:
No

closed 2022-08-31 by Zolko-123