Zolko-123/FreeCAD_Assembly4 #339: How do you select a hole in an imported board with KicadStepUp?

leoheck, 2022-08-30:
How do you select a hole in an imported board with KicadStepUp. 

Kicad Setup Up boards are formed with some sub-assemblies. I can't get a hole from the board itself to create the LCS without having to place it manually. And this is not easy and most of the time the optimal placement is hard to achieve.

I also tried to create a Hole Axis in two ways. First, to use it as the reference for an LCS, in the same Assembly. And alternatively try to use this Hole Axis as the LCS while the imported part is being placed does not work too, since it is not being shown in the LCS list when placing the imported part.

Zolko-123, 2022-08-31:
Issues are for reporting problems, not for general questions or help requests. Please use the FreeCAD forum : 

https://forum.freecadweb.org/index.php

closed 2022-08-31 by Zolko-123

leoheck, 2022-08-31:
Sorry friend. I cannot use the Freecad forum. How can I post there? Not everybody is welcome there.