Zolko-123/FreeCAD_Assembly4 #340: 'FSScrewObject' object has no attribute 'itemText'

leoheck, 2022-09-07:
Adding a screw from Assembly4

[image]

Freecad Version
```
OS: Ubuntu 22.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.30265 (Git) AppImage
Build type: Release
Branch: master
Hash: c35244e74f326a04c5bcab31e984742db582ac78
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * Part-o-magic 1.0.0
  * Part-o-magic.backup1662467294.9039905
  * A2plus 0.4.58
  * Curves 0.5.7
  * fasteners 0.4.4
  * lattice2 1.0.0
  * Assembly4 0.12.3
  * kicadStepUpMod 10.15.1
  * Help 1.0.3
  * ExplodedAssembly
```

Zolko-123, 2022-09-08:
you're right, that's a regression in the new version of fasteners. I have to look

Zolko-123, 2022-09-08:
I released a new version 0.12.4 that should fix the problem. Can you please upgrade and test again ?

leoheck, 2022-09-08:
The screw is being created without issues now.