Zolko-123/FreeCAD_Assembly4 #341: Fasteners not included in BOM

AmeyaPatwardhan, 2022-09-08:
Screws are not added to the BOM. 

Build information: 
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.9, Qt 5.15.4, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.2
Locale: English/India (en_IN)
Installed mods: 
  * fasteners 0.3.44
  * fcgear 1.0.0
  * OpticsWorkbench 1.0.8
  * Assembly3 0.11.3
  * Render 2022.2.0
  * Assembly4 0.12.3

Zolko-123, 2022-09-08:
BOM is currently quite broken, sorry for that

leoheck, 2022-09-14:
This is interesting, I've started to use this WB just because I was thinking it could help me to create a BOM easily. Now, I am seeing this issue here. I found the same behavior while using Asm4.

If you use Fasteners from the Asm4 toolbar, they are inserted directly in the Model instead of being linked as the other Parts and this may be the issue behind it not being listed in the BOM.

@AmeyaPatwardhan If you are looking for a workaround, then you can place the Fastener inside a Part and then link this part to the Model. Then it will be treated the same way as another Part.

See this line in my BOM:
[image]

AmeyaPatwardhan, 2022-09-15:
@leoheck Thanks for pointing this out. It will indeed be very useful.

leoheck, 2022-09-15:
@Zolko-123, I have this Idea.. the fasteners button on Asm4 toolbar could behave this was, instead. It could create a part and place the fastener inside, by default. Then we don't need to keep the Fastener inside the model, and we can also reuse it easily and also it would work with BOM without any extra effort.

Zolko-123, 2022-09-15:
@leoheck : no, the correct way forward is to fix the BOM to include fasteners directly. 

Putting fasteners in a separate Part container and linking it into the Model is a good solution if you're using a group of fasteners, like a screw, a nut, washers, pins ... and then you want to re-use that same group several times.

leoheck, 2022-09-15:
If you have a Part with multiple fatners, then you are saying they are going to be listed individually in the BOM? That is great. I have not tested that myself yet. 

Now, I can't see the difference between having the Fastener added directly to the Model vs it being added inside a Part. I feel, that no one expects the Fastener to be added directly to the Model, by some answers I got from people that tried to use Assm4. This just creates a different workflow. I understand a screw is just another Part.

leoheck, 2022-09-20:
> Putting fasteners in a separate Part container and linking it into the Model is a good solution if you're using a group of fasteners, like a screw, a nut, washers, pins ... and then you want to re-use that same group several times.

In this case, those fasteners are going to appear in the BOM, if the BOM functionality is working fine?

leoheck, 2022-09-21:
@AmeyaPatwardhan this PR https://github.com/Zolko-123/FreeCAD_Assembly4/pull/354 adds fasteners to the BOM. 

I invite you to try it.

leoheck, 2022-09-22:
The previews PR was already merged. 
However, some work was needed.
@AmeyaPatwardhan you can test this one https://github.com/Zolko-123/FreeCAD_Assembly4/pull/355 instead, it may work better. 

@Zolko-123 if you merge this PR then, we can close this issue for now.

leoheck, 2022-09-26:
@Zolko-123 how do I do to differentiate a Part from an Asm4_Part and a Body from the Asm4_Body? Is there a way of doing this?

Zolko-123, 2022-09-26:
@leoheck : you can't they're the same thing. The only difference is that if you use Asm4 to create a Part or a Body, they come with an LCS already. But these LCS are also standard PartDesign::CoordinateSystem

leoheck, 2022-09-26:
I noticed that. 

I was looking for a way to stop the recursion while creating a BOM. To have only important parts in the BOM (Parts and Asm4_Bodies). Since Asm4_Parts may contain other objects that could be added as well. 

I can check for LCS, but as you said they are common objects. 

What if Asm4_Part and Asm4_Body could have a property that they were created with Asm4? Is this possible?

Zolko-123, 2022-09-26:
yes that's possible ... but why ? 

As for recursion, it could be specified by the user when asking for the BoM: we could say to only look for objects directly in the assembly, or 1 level down, or all levels. Or something similar.

leoheck, 2022-09-26:
I would like to have as many levels I can without limitation, this would support sub-assemblies better.

Also, if there is a Part inside the Asm4_Part it means that the user does not care of having the internal objects listed. Because maybe they are just a single Part.

So, lets say this example 1 with

```
Asm4_Part
- Part1
  - Part1.1
  - Part1.2
- Part2
  - Part2.1
     - Part2.1.1
  - Part2.2
```

The BOM would list

1. Part1
2. Part2

Now, suppose this example 2

```
Asm4_Part
- Asm4_Part1
  - Part1.1
  - Part1.2
- Asm4_Part2
  - Asm4_Part2.1
    - Part2.1.1
  - Part2.2
```

The BOM should list, and this for each Asm4_Part regarding the Level

  - Part1.1
  - Part1.2
  - Part2.1.1
  - Part2.2

Actually, the recursion ends when the Part is not Asm4_part

For this to work, an easy way to identify Asm4_Part would be good.
Since you said it is possible, I will test this on my branch to see how it works.
I just have to see a good way to make this property.

Zolko-123, 2022-09-26:
The easiest is to use the existing property "Type". WHich is set to "Assembly" for an Asm4 Model, and empty else. You can set it to `Type=Asm4` and test for this property.

leoheck, 2022-09-29:
The fasteners are included in the development branch. 
This issue can be closed.

closed 2022-10-04 by AmeyaPatwardhan