Zolko-123/FreeCAD_Assembly4 #345: Body: object is not allowedTraceback (most recent call last):

leoheck, 2022-09-10:
I am getting this issue when adding a "New Model" in a file that already has "New Part" models in it. 

I am using parts inside of this file in another file. But now I would like to have a "New Model" here in this current file, assembling them locally. 

[image]

Zolko-123, 2022-09-15:
the problem seems to come from some magic that part-o-magic is doing. I've never used that, I invite you to investigate the issue further down to the core and report back here

leoheck, 2022-09-15:
Ah, really? I am going to remove it, then, to check if this happens again. This WB is too verbose even if I am not using it. I used it to replace a Sketch that was problematic. But I don't use it anymore. It makes a lot of sense. I  going to close this issue, If I see something like this again, I will reopen it, then. Thanks @Zolko-123

closed 2022-09-15 by leoheck