Zolko-123/FreeCAD_Assembly4 #347: There are still something strange with fasteners

leoheck, 2022-09-10:
Working with fasteners, and there is something that is still not working. 

I am using the lates Assm4 0.12.4

- I cannot change the parameters of the fastener while creating it.
- I cannot change its name/label 
- I cannot update parameters while they are linked

[image]

Here, the m4-30 in the objects tree, is the m6 in the parts list.

leoheck, 2022-09-10:
If I put it inside the Part, it shows in the list
[image]

leoheck, 2022-09-10:
Should I place fastners inside a "Part" or should I use them as they are, if adding them from Assembly4 toolbar?

Zolko-123, 2022-09-12:
please use the FreeCAD help forum

closed 2022-09-12 by Zolko-123

leoheck, 2022-09-12:
Are you de developer or just a puppy?