Zolko-123/FreeCAD_Assembly4 #349: Selecting the Configuration

leoheck, 2022-09-10:
This Placed Linked part could have an option to select the Configuration.

[image]

leoheck, 2022-09-13:
> You can try the following: instead of adding a normal part add a variant link, and change the configuration of the linked part of that variant link. This might work, not sure.

Nice, thank you so much for this idea. I will definitely try it. 

> At the time I didn't think about adding Configurations to the possible variants, but this could be a good idea : add a variant link by choosing a particular configuration, and even changing its configuration later.

I am not sure how much of work this requires. But I will also check if I can make something to make this. Also, thanks for letting me know that you think this idea might be useful.

leoheck, 2022-09-29:
An alternative approach would be to include sub-assemblies in the current configuration. In this way, I would be able to update things inside linked Models.

Zolko-123, 2022-09-29:
what would be needed is to be able to apply configurations of inserted parts.

leoheck, 2022-09-29:
Yeah, that would be crazy good. a new level of this tool, actually.
I am thinking that a field listing the configurations in the placePart GUI.
And also a way to specify the configuration of the earch part, in the current assembly configuration to be able to change it eventually.

Zolko-123, 2023-05-29:
Nobody to tackle this issue ?

closed 2023-05-30 by leoheck