Zolko-123/FreeCAD_Assembly4 #351: Adding the LCS in the Model does not end in the right place

leoheck, 2022-09-10:
I am adding an LCS in the Model, to use it when adding it as a Part in another assembly. When I click on a feature in the model (lets say, a circle, it places the LCS where the Part is, and not where the Part is inside the Model. 

It is placing the LCS in the top right of this image, where it should place in the center of the selected circle

[image]

If I turn the part visible, you are going to see that the LCS is where the part is, but not where the part is relative to where it was placed in the Model.

[image]

leoheck, 2022-09-10:
Even using Hole Axis, it is not possible to place an LCS easily in the Model for reusing the Module externally, in another Assembly.

Zolko-123, 2022-09-12:
this is normal behavior, you're not using it correctly, please ask for help in the FreeCAD Help forum

closed 2022-09-12 by Zolko-123

leoheck, 2022-09-12:
I asked, they dont offer any help. How is the correct use of it?