Zolko-123/FreeCAD_Assembly4 #357: Missing Fastners

leoheck, 2022-09-24:
Hi @Zolko-123 there are some fasteners missing in the Asm4 Fasteners button and from the "Change Fasteners parameter"  like spacers, for instance. 

What was the reason behind using Assm4 spacers instead of the ones from Fasteners WB?

This is the BOM listing the one I am using from the Fasteners WB.
[image]

Zolko-123, 2022-09-24:
There are no Asm4 spacers

leoheck, 2022-09-28:
Yes, this is what I am saying. Fasteners WB has Spacers. Asm4 doesn't. Can we have Spacers in the Asm4 too?

Zolko-123, 2022-09-30:
Fasteners is a standalone WB with its own development pace, some of them – the most common ones – are included in Assembly4 WB for convenience. For the others please use the Fasteners WB directly, the resulting parts are usable in Asm4.

closed 2022-09-30 by Zolko-123

leoheck, 2022-09-30:
> For the others please use the Fasteners WB directly, the resulting parts are usable in Asm4.

Ah, that's it then. I thought Asm was doing customization of something. This is good, thanks.

Zolko-123, 2022-09-30:
the only customisations are the colors

leoheck, 2022-09-30:
yeah, and this tells the ures the fasteners are different, which is not true.