Zolko-123/FreeCAD_Assembly4 #358: Part creation changing user's name

leoheck, 2022-09-24:
Hi @Zolko-123 is there any special reason for Asm4 changing the characters the user has put in the Part name?

Please, look the following behavior. I want to create this `name1_name2-name3.name4` Part

[image]

And it ends up being created as `name1_name2_name3_name4`

[image]

I would keep the user's choice unless there is a hard reason for that. But I don't see why it needs to be this way since the user can change the Part name after the creation of it. 

[image]

This is a bit confusing especially when creating a bill of materials. 

Is this a bug or something really important?

leoheck, 2022-09-24:
Ah, ok. The add `addObject` modifie the name to be an internal name, for some reason. 
Then it adds the same as Label.

To keep with what the user wants, it needs something like `newPart.Label = text` after using the `addObject`
I added it here in the line 80. 

[image]

Zolko-123, 2022-09-24:
good catch, thank-you.

closed 2022-09-24 by Zolko-123