Zolko-123/FreeCAD_Assembly4 #361: Fastner config is reseting Z Translation

leoheck, 2022-09-27:
Re-reediting the link placement resets the Y and Z translations.

For instance, I am setting them with 1, here
[image]

When I open the edit mode again, the Y and Z reset to 0 but X remains with the previous value.
[image]

Zolko-123, 2022-09-29:
yes, this is indeed a bug. I need to see

leoheck, 2022-09-29:
I could not check the code yet, but since the X Translation is working, it may not be hard to check against the behavior of the Y an Z translations.

leoheck, 2022-09-29:
Maybe you have to use the Placement property instead of AttachmentOffset (or both)

```
>>> a=App.getDocument('MyFile').getObject('Screw046')

>>> a.AttachmentOffset
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]

>>> a.Placement
Placement [Pos=(-31.0009,-5,4), Yaw-Pitch-Roll=(-0.00172796,0,0)]
````

Zolko-123, 2022-09-29:
nah, it was a problem of the order of initializations. It is fixed now in the development branch. Thank-you for reporting

leoheck, 2022-09-29:
oh, good, it was hard to track it.

leoheck, 2022-10-09:
Hey Zolko, the translation of the screws is still an issue. 

Values are being restored, but there is something wrong now with the X and Y translations. 
Now, it looks like when the Edition opens (or maybe when it ends), the Y gets the X translation. 

I will try to describe what is going on. 

I have this first screw on the left here.
See the Y-translation if it is not too small. 
[image]

But when I do that, it messes up with the X translation
I want it to be close to the other screws on the right side
[image]

Then I edit it again (see the X translation) putting it to where I want, again. 
[image]

Then the Y-translation is totally messed up again
[image]

Then this repeats forever, with the absolute values of the numbers on X and Y translation always increasing

leoheck, 2022-10-09:
Ah, it seems that this issue was already there. 

And it may explain why my screws were in a crazy Y position, to begin with (looking at this Y-axis)

They should not be this far away, since they are close to the center of the initial assembly that I am using as the base to spread open the parts. 

This is the initial picture, after changing the X-translation and spreading the parts, the Y got messed up.
[image]

leoheck, 2022-10-09:
I created a new file to replicate this issue, but I could not replicate it. Even using configurarion. =/ I am not sure what is going on with my file.

leoheck, 2022-10-09:
[Screencast from 10-09-2022 06:27:46 PM.webm](https://user-images.githubusercontent.com/1277920/194780251-76d4e915-a375-4166-9c5e-48d88c47813c.webm)

Ah, this is not an issue at all. The hole axis is at a different angle, not perpendicular to X and Y axis, so here you see how the screw move. 
This is something that I kind of think is hard to fix. Sometimes my LCS are rotated and I don't know the best way to rotate them to the angle I want automatically by tweaking the LCS parameters.

closed 2022-10-09 by leoheck