Zolko-123/FreeCAD_Assembly4 #367: Unkown bug

leoheck, 2022-10-02:
I am still reviewing the BOM thing but I found a little bug. Not sure what it is but I am not touching this file. 

```
➜  Assembly4 git:(development) ✗ Unhandled unknown exception caught in GUIApplication::notify.
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 129, in IsActive
    def IsActive(self):
<class 'KeyboardInterrupt'>: <unknown exception data>

```

leoheck, 2022-10-02:
Feel free to close this if it is not important, I am just sharing because it happened now, all of the sudden.

Zolko-123, 2022-10-27:
If there are no news on this I close this Issue

leoheck, 2022-10-27:
Yeah, it did not happen again.

closed 2022-10-27 by leoheck