Zolko-123/FreeCAD_Assembly4 #369: New features

leoheck, 2022-10-04:
Hey @Zolko-123, I would like to implement these functions in Assembly 4. Do you accept them as a good addition?

- List Linked files in Console (List and Tree Format) in the same Button
- Export the Assembly files to a folder, o maybe to a zip file

Zolko-123, 2022-10-04:
> List Linked files in Console (List and Tree Format) in the same Button

yes, that's good, but we had something similar already, you should check whether that helps:

```
def on_button_list_parts(self):            
    self.level = 0
    for obj in App.ActiveDocument.RootObjects:
        if str(obj) == "<Part object>":
            print("Level " + str(self.level) + " Top", str(obj.Label))
            self.find_sub(obj)

def find_sub(self,obj):
    self.level += 1            
    for subobj in obj.OutList:
        if str(subobj) != "<App::Origin object>":
            if str(subobj) == "<Part object>":
                self.space = 2 + self.level
                App.Console.PrintWarning("Level " + str(self.level ) +", " * self.space + str(subobj.Label)+"\n")
                self.find_sub(subobj)
    self.level -= 1        
```

also, look at the Macro "TreeToAscii" . If I remember right the output was in a text window which is better than the console. 

> Export the Assembly files to a folder, o maybe to a zip file

that would be very good, yes. SolidWorks has this functionality which is very practical. It might by quite difficult though.

leoheck, 2022-10-04:
Yesterday I made this function here list/tree. Posted it here.
Personally, I feel it looks quite good. 
Right now it showing the result in the console
https://forum.freecadweb.org/viewtopic.php?f=20&t=34806&start=1140#p630370

The code is here (Gist on Github)
https://gist.github.com/leoheck/0933ca63db54e2f7f22ada0e2a4f811b

>> Export the Assembly files to a folder, o maybe to a zip file

> that would be very good, yes. SolidWorks has this functionality which is very practical. It might by quite difficult though.

I was thinking this would be easy. Why do you think this would be difficult?

Zolko-123, 2022-10-04:
> Why do you think this would be difficult?

only a feeling. Might be wrong though. This might be the case of not seeing the forest because of the tree

leoheck, 2022-10-05:
I made a quick attempt on this, and here it is.
https://github.com/Zolko-123/FreeCAD_Assembly4/pull/370

There are 2 buttons to list files, currently in the console (a list of unique and absolute filenames and a tree of relative file paths)
There is another button to export a zip, in the current folder. The limitation here, to preserve the files' structure and keep the assembly working when unpacked is to keep files under the main assembly folder. It is currently not exporting the package if there are external files.

leoheck, 2022-10-10:
@Zolko-123 do you have a code to share that launches a file browser? 

Right now I am hardcoding saving the exported zip in the same directory as the assembly file, but it would be better to present the file browser so the user can tell where it wants the file and its name.

Zolko-123, 2022-10-10:
look in _AnimationExportLib_

Zolko-123, 2023-04-10:
See #410

closed 2023-04-10 by Zolko-123