Zolko-123/FreeCAD_Assembly4 #374: Problem with auto-numbering in import part

abetis, 2022-10-15:
Hi,

I preparing a PR with a change, but want to make sure there's no problem with the proposed solution.

Current code does the following:
- Check if the last character of the added part label is a number => Assume that numbers after "_" character are from auto-numbering logic
- If there is no "_" in the name => Assume last 3 characters are from auto-numbering logic

I have parts (imported from STEP files) from a vendor that have part numbers in the format of 1111-2222-3333 (3x 4digit numbers separated with dashes)
Looks like the links logic doesn't like the dash ("-") and convert it to the underline ("_"), so I end up with the 1111_2222_3333 naming for the imported parts.
The auto-numbering logic strips the last digits since there is "_" in the Label and let Asm.nextInstance to decide what Label should it be, that lead to a mess in the naming and in some cases differences between the selected Label and assigned internal name.

--> Also, there is a different behavior if I select a part and press the Import Part button, or have no selection, press open the Import Part dialog and select the part.

My proposal:
Auto-numbering should be added as a suffix in the format of "/123", starting with "/001" even for the first and a single item added.
The internal naming is ok with the "/", it's converted to "_" as all other "problematic" characters.
All the logic should be concentrated in the Asm.nextInstance, so all the places in the code will have the same behavior.

Zolko-123, 2022-10-16:
Although you have correctly described how this works, I'm not convinced. This is for auto-numbering, if a user has special needs he can override the default name proposal.

abetis, 2022-10-16:
Right now current auto-numbering breaks all my parts naming.
And I don't think a user wants to mess with the auto-numbering.
The user might definitely change a part name if needed, even if its
derived from the same part file.

I'll push the changes later, please take a look on it after that.

abetis, 2022-10-16:
Please take a look: #375 

The main problem is that current code removes something from the name that it didn't add, so legit names are destroyed.
Specifically those including "_" and a number at the end of the name. All the files I use have that numbering convention.
All goBILDA catalog numbers are in "1111-2222-3333" format...

Zolko-123, 2022-10-16:
auto-naming is only a proposal, it's a helper function. User can override it easily. I'm sorry if it doesn't help in your use-case.

abetis, 2022-10-17:
Can I turn it OFF? probably NO.
Right now the code strips off automatically my valid model names, without any warning.
Having labels in the assembly tree equal to the internal names helps alot when you point on something in the 3D view and look on the name at the bottom of the application and use it as a decision what model should be used to connect the new imported model to.
Having the same instance of a model with different naming because the code decided to strip the last part of the name and add there some other numbers doesn't help and confise.

From the beginning of the Asm4 development I see that we use it from different ways:
You "create" parts inside the FreeCAD, I import them from STEP files and use only the assembly module.
You can name the parts whatever way you want, I'm more restricted to the way the manufacturer name the models in their STEP files. Changing the default names for hundreds of ready models is not the right way to do.

If you'll try to use my proposed code the way I work, with importing many similar parts to the assembly, you'll understand that current code doesn't do it properly. Especially when I'll have to export BOM based on the naming of the models that were changed automatically.
Yes, I can change the names one by one, but what for did you add the auto-numbering? To help and save the time? Or to confuse the user and make his life harder?
Those are the parts I use, if you want to try and assemble something using those parts with and without my PR. Just add several parts with the same name to the assembly and see...
https://www.gobilda.com/structure/

abetis, 2022-10-17:
Adding more: the naming of the parts include their length, such as "1120-0003-0096" is a 96mm length and "1120-0006-0168" is 168mm channel from the 1120 series.
That part of the name is stripped by the auto-numbering...

Zolko-123, 2022-10-17:
> You "create" parts inside the FreeCAD, I import them from STEP files and use only the assembly module.

that's what is done in the first tutorial, did you check it :

https://github.com/Zolko-123/FreeCAD_Examples/blob/master/Asm4_Tutorial1/README.md#document-bearing

> what for did you add the auto-numbering?

because if you add a part 5 times, i.e. you have 5 instances of the same part, then it's logical to call them _Part_1_, _Part_2_, _Part_3_ .... But you might also want to call them _Part_left_ and _Part_right_, in which case the user must override the default proposed name. 

How do you intend to call multiple instances of a part called _111_444_196_ ?

abetis, 2022-10-17:
I saw that tutorial, looks like it was created before the auto-numbering code was added since the dialog says that the entered name must be unique in the dialog tree...
When you add 2-3 bearings to a 4-5 total parts in the assembly, it's obviously easy to change the name manually.
When you add 30-40 parts to the assembly, remove some, swap to other models since you understand that those will suit the needs better, you don't really remember if you already have part "1120-0003-0096" in the assembly and how many of them...
That's where you need the auto numbering logic. I just want to add the "1120-0003-0096" to the assembly, if there are 4 of those already, it should be auto-numbered as 5'th and not as "1120-0003002" how the current code decides to name it.

The code in the PR adds "/001", "/002" and so on to all the added models.
I can add a checkbox to the dialog that will not apply the auto-numbering for those who want to have only "LeftPart" or "RightPart" in the list. But I don't see much harm appearing it as "LeftPart/001" and "RightPart/001" in the tree.

Believe me that I wont spend my time on something "silly" if that wont fix something and make my work more productive.
I'm in a process of creating an assembly with hundreds of those parts and need a more logical solution than the current auto-numbering code does.

Zolko-123, 2022-10-17:
> Believe me ...

I never believe anyone saying "believe me" . And you didn't answer my question:

How do you intend to call multiple instances of a part called _111_444_196_ ?

abetis, 2022-10-17:
> How do you intend to call multiple instances of a part called _111_444_196_ ?

I answered it several times:
The code in the PR adds "/001", "/002" and so on to all the added models.

So when you add part "111_444_196" several times, the naming will be: "111_444_196/001", "111_444_196/002", "111_444_196/003" and so on.
And when you reach 999 it will continue to "/1000", "/1001" and so on. Something that current code won't do since it strips only 3 numbers after the "_".

Lets get productive:
Did you try to add few modules with the naming of "1111-2222-3333" to the assembly and see how auto-numbering behave?
You can use the bearing from here for the same tutorial you mentioned:
https://www.gobilda.com/1600-series-non-flanged-ball-bearing-8mm-id-x-22mm-od-7mm-thickness-2-pack/
But it's enough just to add them to the assembly without actually attaching them to anything.

Do we want the auto-numbering behavior? I say yes.
Should it change valid names as part of its logic? I say no.

Do you have other behavior proposal besides leaving it as is?

Zolko-123, 2022-10-17:
> The code in the PR adds "/001", "/002" and so on to all the added models.

even if a part is only used once ? That's not acceptable, sorry. Many parts are used only once, and some instances are named explicitly like _left_ and _right_ or _front_ and _back_, or similar. 

But I can agree to use something else than "_" as separator for instance naming. For example, if a user names an instance _part/1_ (or _111_333_456/1_) then we can increment it as _part/2_ , _part/3_ ...

abetis, 2022-10-17:
When you add the "111-333-456" part for the very first time, the code doesn't know if it will be a single model or there will be 5 more of them in the assembly.

How about adding a checkbox saying something like "Apply auto-numbering" and some explanation that it will add "/001, /002" and so on? I'm ok to have that checkbox OFF by default and save its value during the run-time of FreeCAD. So if I set the checkbox, it will stay ON until I close FreeCAD and reopen it again.
The dialog will check that the supplied name is not a duplicate in the document. For auto-numbered names it will never happen, but for manually added it's a good practice to check.

The "/" only shown in the tree, the internal name will convert it to "_" in any case and the check should be done against "_" version in the name. Model with label "1102-0005-0040" is added to the assembly as "1102_0005_0040" for some reason, which is also pretty annoying.

abetis, 2022-10-17:
I don't really understand what do you propose to do.
The checkbox is always there, you can check or uncheck it for each part you add.
If I want auto-numbering, I'll check it once and it will stay checked for the next added models.
If you don't want auto-numbering, don't check it at all.

When you add models, delete models, swap models in the middle of a large assembly, you don't really know if the model you add already exist in the assembly or not. Checking the tree for every part you add is a huge waste of time.

Also, as I wrote, the internal naming don't allow "/" in the name, and not "-" as well. It converts all those characters to underscore. So there is no way to change the scheme to "/X". It might appear in the label, but not in the internal naming.
When you use the App.ActiveDocument.getObject() to check if there is such object, you have to check against the underscore version and not the "/X" version. That's the reason why the code strips the end of the valid "1111-2222-3333" name.

Looking manually in the tree to check if the model was already in there is not the right way to go.

Zolko-123, 2022-10-17:
> I don't really understand what do you propose to do.

easy: nothing. But if you propose something, I'll look into it.

abetis, 2022-10-17:
Man, I don't understand this discussion at all.
I report that there is a problem when you use legit valid model names in format "1111-2222-3333" and propose a solution.
Possible that the solution needs some tweaks since everybody use the module in different ways.
For now it sounds like you don't really want to solve anything and it's "silly" for you to change anything, despite the fact that auto-numbering doesn't work properly for me.
I bet you didn't even try to check it and we ping-pong the comments for the second day already...

Zolko-123, 2022-10-27:
See discussion on the forum:

https://forum.freecadweb.org/viewtopic.php?p=634905#p634905

closed 2022-10-27 by Zolko-123