Zolko-123/FreeCAD_Assembly4 #380: Variables in parts are unmanageable

gergaly, 2022-10-25:
Hi Zolko,

I've just started using this Assembly4 workbench recently and I really like it. However I run into a small problem. When I copy/paste a part from different document into the Parts section of the assembly and the part has variables I can't manage them anymore. In the assembly document Add/Del variable GUI command works only on the Variables of the assembly model.

Reproduction:
New document, part, body, create something,
Assembly4 WB, add variable
Save
New document, new assembly4 model
copy/paste part from previous document
Now the assembly4 document has two Variables section. One, named "Variables" in the model. The other one, named "Variables001" is in the part with existing variable in it.
The variables in "Variables001" cannot be deleted. I can't add new variables here. However I am still able to use these variables in expressions.
[image]

Looking into the code it seems like you are only looking for 'Variables' object in the Active Document. Can this be done in the Active Part as well? I don't know how this supposed to work but I really like the idea of having variables in the part level as well as in the model level.
```
OS: Ubuntu 18.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * toSketch 1.0.1
  * Assembly4 0.12.4
  * Part-o-magic 1.0.0
  * fasteners 0.4.16
```

gergaly, 2022-10-26:
> > When I copy/paste a part from different document into the Parts section of the assembly
> 
> what is your reason for doing this ? Why don't you link (insert) the original Part directly into the assembly ?

When I link a part from an other document I have to keep all linked documents together. To be able to open the assembly document again I need all the files with all the parts, correct? I'd prefer the all my parts would live in the same document, so I don't have to keep track all the files just the one.

But anyway what is the "Parts" container for?

Zolko-123, 2022-10-26:
The Parts container group is to make assemblies in 1 file (document). It's useful for small stand-alone assemblies. 

> To be able to open the assembly document again I need all the files with all the parts, correct ?

What do you mean by that ?

Zolko-123, 2022-10-26:
If you open an assembly which has a part inserted that comes from another document, then FreeCAD will automatically open that other document in the background. The path to that other document is stored in the assembly document, therefore if you move the target part to another directory in your computer, then it will not be found.

It's actually as simple as possible, and identical to what other CAD systems do

gergaly, 2022-10-26:
> The Parts container group is to make assemblies in 1 file (document). It's useful for small stand-alone assemblies.

So, we can have parts in the Parts container in the assembly4 document. And I've shown we can have Variables in these parts as well. Now we can discuss the OP: I can't add or delete these type of variables.

In the meantime I've written these python functions to do the trick for me:
```python
def findP():
  for Sel in Gui.Selection.getSelection():
    if Sel.TypeId == 'App::Part':
      for P in Sel.Group:
        if P.TypeId == 'App::FeaturePython' and P.Type == 'App::PropertyContainer':
          return P
  return None

def addV(VName,VType):
  P = findP()
  if P != None:
    return P.addProperty('App::Property'+VType, VName, 'Variables' )

def remV(VName):
  P = findP()
  if P != None:
    return P.removeProperty(VName)
```
I just paste them in the python console and I can add/remove variables in the selected part until my nose bleeds.

gergaly, 2022-10-28:
Hi @Zolko-123,
Since no one else has noticed this behaviour and I've found my workaround you can forget about it.

Zolko-123, 2022-10-28:
thank-you for your comments, and yes, I've mixed feelings about this. So if your satisfied with your solution, can you please close this issue ?

gergaly, 2022-10-28:
yes, close the issue please. I've learned a few things since I've opened it.
Udv

closed 2022-10-28 by Zolko-123