Zolko-123/FreeCAD_Assembly4 #381: Feature request: explode aid

gergaly, 2022-10-27:
Hi,

Please consider the following feature request: Built-in aid to exploded view.

How I did it:
I've added a new variable called 'explode' to the model. Then I've modified every attachment's position along one axis (this is the direction where the two parts were to move from each other), in the Formula editor I added/subtracted 'Variables.explode' to/from the existing value. So changing/animating the explode variable would explode the assembly.

How would I imagine this in the Assembly4 WB:
Upon inserting a new part, in the 'Place linked Part' dialog the user would be able to check one or multiple axis checkbox to be included in the explode. And the position of the attachment along this axis would include 'Variables.explode' by default.

Thank you for your consideration
[image]
Github didn't let me attach the FreeCAD document. I've renamed it to zip and uploaded like that.[Assembly4_explode.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/9882342/Assembly4_explode.zip)

Zolko-123, 2022-10-27:
This is actually a good idea, but since you were able to do it with the current code I don't see what should be added/changed in the WB. What would be more useful is to document your approach so that other can duplicate it.

gergaly, 2022-10-27:
> This is actually a good idea, but since you were able to do it with the current code I don't see what should be added/changed in the WB.

Well, it would be a lot easier to tick a checkbox during part insert than to edit the attachment position after part insert.
> What would be more useful is to document your approach so that other can duplicate it.

I'd be happy to do that. Let me know how should I do it and where to put it and I'll write it up.

luzpaz, 2022-12-21:
any traction on this ?

Zolko-123, 2022-12-22:
@luzpaz :

> any traction on this ?

While I think it's generally a good idea, I also think that this should not be over-engineered. I didn't find a good way to do this yet (put a checkbox on each part's attachment on each direction X-Y-Z could be a solution, but it only works if the user does it for all parts)

gergaly, 2022-12-22:
Since the OP I've been applying this to my assemblies and I've came to the conclusion that a checkbox for each axis wouldn't be enough. Sometimes I have to use the explode value in the same axis but with different direction. And also I've had drawings where I had to multiply the explode value to be able to reach the desired goal. So I think the solution would be an edit box for each axis where the user could enter the explode multiplier value: 1 for normal, -1 for reverse direction or any other number if desired.

Zolko-123, 2022-12-23:
> So I think the solution would be an edit box for each axis

this means a dedicated function then. The current object placement seems so far quite good (at least nobody complains) therefore I wouldn't overload it since most users won't use this dynamic exploded assembly function. Also, there are already _Configurations_ to do something similar, even though not dynamic. 

So the solution would be to call a function, which would launch a Task panel that lists all the objects, and the user could select for each object the axis and the value of the explosion. This would then create a variable that would be added to each selected object's. Ideally, once a good exploded state is found, the function can create a corresponding Configuration so that same state can be called again. 

Not completely trivial.

Zolko-123, 2023-04-11:
Not a current priority

closed 2023-04-11 by Zolko-123