Zolko-123/FreeCAD_Assembly4 #382: Cannot enter more than 2 decimals in the variables dialog box

Jeanmimi0168, 2022-12-06:
Variable creation dialog box doesn't allow to enter more than 2 decimals, but in the property list you can enter more than two decimals.

Version 0.12.4 08/11/2022

Jeanmimi0168, 2022-12-06:
Seems to be due to default value of QDoubleSpinBox.decimals that could be changed by QDoubleSpinBox.setDecimals(int).

Zolko-123, 2022-12-07:
good catch: do you know how to do a PR for that ?

Jeanmimi0168, 2022-12-07:
Tried it, tell me if I made it right.
Just CAD user with some background of python.

Zolko-123, 2022-12-07:
looks good. But can you please make the PR against the development branch ? Also, 6 digits ... you're planning ahead !

Jeanmimi0168, 2022-12-07:
I needed to scale about a bitmap measurement, and there was more than that!
I try to change the PR to dev branch

Jeanmimi0168, 2022-12-07:
Did not found how to change branch, just closed the PR and opened a new one on dev branch.

Zolko-123, 2022-12-07:
thnaks, can-you please close this issue now ?

closed 2022-12-08 by Jeanmimi0168