Zolko-123/FreeCAD_Assembly4 #393: Most of the time, ASM4 does not Show LCSs of Linked Bodies

leoheck, 2023-01-26:
Sometimes I can see items inside the "Select LCS in Part" sometimes, most of the time, I can see nothing there.  I am unsure what is going on, but the LCS list is empty. 

I was about to report this hours ago, but then suddenly it started working.. but after some time, the LCS stopped being displayed to place one part. 

For instance, these are my objects
[image]

This is what I am getting, most of the time. An empty list.
[image]

Project created today using
```
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31685 (Git) Snap 566
Build type: Release
Branch: master
Hash: b853278296ee93dd08a22557c2a44f977efa9dd0
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.5
  * Part-o-magic 1.0.0
  * freecad.gears 1.0.0
```

Zolko-123, 2023-01-26:
very strange. You seem to be using  a Snap version, could you please test with an AppImage and see whether that happens also ?

Zolko-123, 2023-01-26:
There is something wrong with that model: the LCS aren't shown in the 3D window either. I tried adding anolther LCS to those bodies and they aren't visible either. 

When I create another Body, in Asm4 or PartDesign, then the LCS it contains are shown in the 3D window and in the LCS list.

How did you create those bodies ?

Zolko-123, 2023-01-26:
Try this : select the body, go to the _Property View_, there go to the _View_ tab, and in the _Base_ section look at the _Display Mode_, and choose "_Through_" instead of "_Tip_".

leoheck, 2023-01-27:
> How did you create those bodies ? 

Using the snap version of freecad and the toolbar buttons of asm4. 

I Will try tô remove and reinstall the asm4, too.

Zolko-123, 2023-01-30:
There is a problem in your file: the _Display Mode_ of the bodies that fail is set to _Tip_ instead of _Through_ . Nothing to do with Asm4