Zolko-123/FreeCAD_Assembly4 #395: BOM issues in 0.12.5

juddidier, 2023-02-03:
Hi,
I tried the "new" BOM (from December) and the script stumbled several times. I dug into the code and found followings:
First error:
`14:50:15  ASM4> BOM of the legacy Assembly 4 Model`
`14:50:15  ASM4> BOM following sub-assemblies`
`14:50:15  ASM4> [1]    | App::Link | assembly#Part | Part`
`14:50:15  Running the Python command 'Asm4_makeBOM' failed:`
`Traceback (most recent call last):`
`  File "C:\Users\tweety\AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 116, in Activated
    self.listParts(self.model)`
`  File "C:\Users\tweety\AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 421, in listParts
    self.listParts(subobj, level+1, parent=obj)`
`  File "C:\Users\tweety\AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 160, in listParts
    self.listParts(obj.LinkedObject, level + 1, parent=obj)`
`  File "C:\Users\tweety\AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 253, in listParts
    if self.infoKeysUser.get("Part_Label").get('active'):`
`'NoneType' object has no attribute 'get'0 Selection.cpp(1674): Object not found`

In the script you use 5 code sequences that are quite similar (but not same) described as "Recover the record" (lines 168ff, 178ff, 243ff, 253ff, 312ff). The script hangs at the if clause that is outside of the try-except.
IMHO the easiest and safest way would be to apply at all the positions (including the if-clause into the try and not nesting a try into a try):
`try:`
`   if ....`
`     command`
`except:`
`   command`

In addition, think in the "Standalone Model-Partdesign" (lines 309ff) the "recovery" of obj_label is missing completly

Second error, the code hangs is line 298 (in my case it happens at line 303 since I had to modify my code according to above to continue testing):
`15:13:00  ASM4> BOM of the legacy Assembly 4 Model`
`15:13:00  ASM4> BOM following sub-assemblies`
`15:13:00  ASM4> [1]    | App::Link | assembly#Part | Part`
`15:13:00  ASM4> [2]        | 1x | App::Part | Element1#Part | Part`
`15:13:00  Running the Python command 'Asm4_makeBOM' failed:`
`Traceback (most recent call last):`
`  File "C:\Users\tweety\AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 116, in Activated
    self.listParts(self.model)`
`  File "C:\Users\tweety\AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 426, in listParts
    self.listParts(subobj, level+1, parent=obj)`
`  File "C:\Users\tweety\AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 160, in listParts
    self.listParts(obj.LinkedObject, level + 1, parent=obj)`
`  File "C:\Users\tweety\AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 303, in listParts
    self.Verbose += "- " + prop + ": " + data + " " + info + "\n"`
`can only concatenate str (not "App.Document") to str1e-07 Selection.cpp(1674): Object not found`
 
I divided the line into 3 separate commands and it seems the problems is the variable _data_. To my understanding the command **getattr** (line 287) defines the variable _data_ not as string but something else, and therefore the string concatenation in line 298 fails. Since data seems to be a some type of class, the code in line 298 should dig deeper into the class-hierarchy.

Same applies to line 228.

In addition, wouldn't it be more simple to replace line 279 with lines 286-289? The sole purpose of line 279 is to eventually fire an exception. Why not also fill the variable _data_ at the same time?

Regards
DidierJ

juddidier, 2023-02-03:
Oh, I have forgotten, in line 356 the if clause is probably not correct.
You wrote (simplified):
`elif a == b and (c or d) > -1`
If you have a Featurepython (the a==b condition) element in your assembly the if clause always fire, even if it's not a fastener (I tried with a VariantLink).
I think the correct behaviour should be:
`elif a==b and (c >-1 or d>-1)`
or fully written:
`elif obj.TypeId == 'Part::FeaturePython' and (obj.Content.find("FastenersCmd") > -1 or obj.Content.find("PCBStandoff") > -1):`

Zolko-123, 2023-02-06:
The BoM has been written and re-written by many people to suit their specific needs. Since I don't use it really, it's kinda broken. If you know how to make a good BoM for most use-cases (cut-list, assemblies, part-list ...) then you're welcome.

Actually, if you have some available time, the best would be not to dive into the code (which is a mess) but to write down what we ultimately want, and then write the corresponding code.

One of the things that I'd like is to have templates for _PartInfo_. This way, a project or company can make pre-defined  templates with usual properties. And this _PartInfo_ should work together with `makeBOM()`. The BoM itself could have templates for different users

juddidier, 2023-02-06:
> One of the things that I'd like is to have templates for _PartInfo_. This way, a project or company can make pre-defined templates with usual properties. And this _PartInfo_ should work together with `makeBOM()`. The BoM itself could have templates for different users

That would be for sure a very good solution.

>Actually, if you have some available time, the best would be not to dive into the code (which is a mess) but to write down what we ultimately want, and then write the corresponding code.

I don't think I could write the code from scratch. Adding here and there an if or similar is not big deal (error correction), but doing the whole stuff, well... I haven't dived into the object model of Freecad. There are quite a few sequences in the BOM script where I'm absolutely lost.

leoheck, 2023-02-09:
Hi @juddidier  you can try my fork of this repo, the BOM is working there. The dev branch here was updated so it may not work as my fork anymore. Cheers.

juddidier, 2023-02-09:
Thanks leoheck, I'll give it a try.

juddidier, 2023-02-10:
> Hi @juddidier you can try my fork of this repo, the BOM is working there. The dev branch here was updated so it may not work as my fork anymore. Cheers.

sorry, additional question: which branch is the actual one? linked list files or development?

leoheck, 2023-02-10:
Both branches will work.

Anytree can be installed with pip with this command:
```bash
pip install anytree
```

BuhtigithuB, 2023-02-20:
Hi guys,

Stumble on the BOM issue here : https://forum.freecad.org/viewtopic.php?t=75567

I guess this issue is related...

Can we revert to a working version of the feature?

Thanks

leoheck, 2023-02-22:
My BOM as a PR with one latest Fix.
Please try it and report anything that is ugly for you guys. 
Cheers.
https://github.com/Zolko-123/FreeCAD_Assembly4/pull/403

leoheck, 2023-02-23:
@juddidier can you try the development branch of this repo? my PR was merged.

theblackraven, 2023-04-22:
I get this error message, when using the BOM feature: using the makeBomCmd from developing or 0.50.1
... AppData\Roaming\FreeCAD\Mod\Assembly4\.\makeBomCmd.py", line 264, in listParts
    if obj_label == "Model":

leoheck, 2023-04-23:
The Model was renamed to Assembly. You can rename your Assembly to Model, to use it right now, you could also edit that line in the code to be "Assembly". But it would be better to have this test changed to something more robust in the future since users can rename the Model/Assembly to anything they wants to call it.

einhander, 2023-05-06:
My Assembly called Model, but I have following error:
```
SM4> Listing linked files in a tree
18:18:54  Water-Dacha.FCStd
18:18:54  ASM4> Listing ended
18:19:48  ASM4> BOM of the legacy Assembly 4 Model
18:19:48  ASM4> BOM following sub-assemblies
18:19:48  ASM4> [1]    | App::Link | Water_Dacha#Pipe32_300 | Pipe32-1501
18:19:48  ASM4> [2]        1x | PartDesign::Body | Water_Dacha#Body003 | Pipe32-300
18:19:48  ASM4> [1]    | App::Link | Water_Dacha#Pipe32_100 | Pipe32-1502
18:19:48  ASM4> [2]        1x | PartDesign::Body | Water_Dacha#Body005 | Pipe32-100
18:19:48  ASM4> [1]    | App::Link | Water_Dacha#Socket001 | Socket001
18:19:48  Running the Python command 'Asm4_makeBOM' failed:
Traceback (most recent call last):
  File "/home/einhander/.local/share/FreeCAD/Mod/Assembly4/./makeBomCmd.py", line 116, in Activated
    self.listParts(self.model)
  File "/home/einhander/.local/share/FreeCAD/Mod/Assembly4/./makeBomCmd.py", line 428, in listParts
    self.listParts(subobj, level+1, parent=obj)
  File "/home/einhander/.local/share/FreeCAD/Mod/Assembly4/./makeBomCmd.py", line 160, in listParts
    self.listParts(obj.LinkedObject, level + 1, parent=obj)
  File "/home/einhander/.local/share/FreeCAD/Mod/Assembly4/./makeBomCmd.py", line 264, in listParts
    if obj_label == "Model":
       ^^^^^^^^^

cannot access local variable 'obj_label' where it is not associated with a value
````

Zolko-123, 2023-05-06:
> But it would be better to have this test changed to something more robust

that function exists: it is `getAssembly()` in Asm4_libs, and can be called with `Asm4.getAssembly()` which returns a valid Asm4 assembly (even older _Model_ objects) if it exists, or None. Use it as:

```
assy = Asm4.getAssembly()
if assy:
    # it's a valid Asm4 assembly
else:
    # it's not
```

einhander, 2023-05-06:
> My Assembly called Model, but I have following error:

Upd. An error occurs if my Assembly has part from Assembly Workbench, parts from Part Design doesn't cause the error.

snelweg, 2023-05-17:
I found this https://github.com/Ediforce44/FreeCAD-Assembly4-BOM looks promising. At least it runs in my FC version and gives some output.

`[code]
OS: Ubuntu Core 20 (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33216 (Git) Snap 708
Build type: Release
Branch: master
Hash: ea35f0f003a68730b21819372ed34bcc4a13e482
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * dodo 1.0.0
  * Render 2023.2.5
  * fasteners 0.4.56
  * FeedsAndSpeeds 0.5.0
  * Dracula 0.0.4
  * Defeaturing 1.2.0
  * POV-Ray-Rendering
  * Behave-Dark-Colors 0.1.1
  * Assembly4 0.50.2
[/code]
`

leoheck, 2023-05-17:
@snelweg what is different than what we have here in this repo?

Can you summarize the problem being discussed here?

snelweg, 2023-05-17:
> @snelweg what is different than what we have here in this repo?

Well for starters, your fork has the same problems as the Zolko one when I attempt to make a BOM. It does not work (for me).

There are error messages :+1: 
`Running the Python command 'Asm4_makeLocalBOM' failed:
Traceback (most recent call last):
  File "/home/ton/snap/freecad/common/Mod/FreeCAD_Assembly4/./makeBomCmd.py", line 116, in Activated
    self.listParts(self.model)
  File "/home/ton/snap/freecad/common/Mod/FreeCAD_Assembly4/./makeBomCmd.py", line 421, in listParts
    self.listParts(subobj, level+1, parent=obj)
  File "/home/ton/snap/freecad/common/Mod/FreeCAD_Assembly4/./makeBomCmd.py", line 160, in listParts
    self.listParts(obj.LinkedObject, level + 1, parent=obj)
  File "/home/ton/snap/freecad/common/Mod/FreeCAD_Assembly4/./makeBomCmd.py", line 253, in listParts
    if self.infoKeysUser.get("Part_Label").get('active'):

'NoneType' object has no attribute 'get'` 

> Can you summarize the problem being discussed here?

The same problems as the Zolko branch. The assembly  branch I mentioned at least works for me and exports nicely to a csv file.
I do hope that soon there will be a nice BOM functionality. Yours looks promising and I would like to try it out when it works.

leoheck, 2023-05-17:
> It does not work (for me).

Cool, what it is doing if not working? Can you describe it briefly for me?

Do you have a project to share so I can check what is going on? I was working on my projects in the past. But you may have something different or are using a different version of Asm4 that has introduced bugs or something.

Also, can you edit your answer by formatting the terminal output with the "code' format so I can read it easily? 

After that, I can give it a look to see how to improve the BOM.

leoheck, 2023-05-22:
Hey guys, I am trying to improve this BOM one more time by adding a GUI, so users can tweak some settings. Some code has changed since the last time I worked on it, so it takes time to understand and use what we have.

The current GUI I have is this one:

[image]

My first effort was to build this GUI, then I am managing to have a better well-organized way of walking through the objects tree. And now I am trying to understand how this usersInfoKey has evolved since I am not being able to get the info from the parts right. 

If you have any ideas or requests, please let me know, and I will try to add them to the GUI/BOM

Right now, the initial set of rules it more or less like this:

```
    >>>> (Initial) BOM Rules <<<<<
    
    - Asm4 App:Part
      IF Main/Root Asm4: Follow objects
      ELSE:
         IF Visible:
            IF follow_subassemblies: Follow objects on Group
            ELSE: Create/Update its Record
    
    - App:Part (not Asm4)
      IF nested_level < max_nested_level: Follow objects on Group
      ELSE: Create/Update its Record
    
    - Visible App::Link
      Follow ALL LinkedObjects (passing the ElementsCount)
    
    - PartDesign::Body
      Create/Update its Record
    
    - Visible Python::Array
      Follow objects on Group (passing the ElementsCount)
    
    - Visible Python::Fastener
      Create/Update its Record
    
    - Visible App::DocumentObjectGroup
      Follow objects on Group
```

leoheck, 2023-05-22:
Hi, I am working to improve this makeBOM. 

I am building a GUI this time (since I've learned how to do it). So the user will have some items to tweak the output.

This is the current GUI:
[image]

The current set of rules for Parsing Objects is this:

```
    >>>> (Initial) BOM Rules <<<<<
    
    - Asm4 App:Part
      IF Main/Root Asm4: Follow objects
      ELSE:
         IF Visible:
            IF follow_subassemblies: Follow objects on Group
            ELSE: Create/Update its Record
    
    - App:Part (not Asm4)
      IF nested_level < max_nested_level: Follow objects on Group
      ELSE: Create/Update its Record
    
    - Visible App::Link
      Follow ALL LinkedObjects (passing the ElementsCount)
    
    - PartDesign::Body
      Create/Update its Record
    
    - Visible Python::Array
      Follow objects on Group (passing the ElementsCount)
    
    - Visible Python::Fastener
      Create/Update its Record
    
    - Visible App::DocumentObjectGroup
      Follow objects on Group
```

If you have ideas on the GUI/Rules or any other requests, let me know.
Right now I am working to understand and use the userInfoKeys that has changed since the last time I worked on this. In other words, the parsing of the object's data (Part, Body, Fasteners, and custom user settings). This is still not working for me.

leoheck, 2023-05-23:
Here in this branch is the initial work, if someone wants to check it out.
https://github.com/leoheck/FreeCAD_Assembly4/tree/bom-improvements-with-gui

snelweg, 2023-05-23:
> Here in this branch is the initial work, if someone wants to check it out. https://github.com/leoheck/FreeCAD_Assembly4/tree/bom-improvements-with-gui

Great! Will have a look this weekend. Thanks!

leoheck, 2023-05-26:
@snelweg I added some fixes and improvements, please, update your repo.

leoheck, 2023-05-27:
Added a progress bar which is good for big projects.. 

[image]

snelweg, 2023-05-27:
> @snelweg I added some fixes and improvements, please, update your repo.

Thanks, will try it.

snelweg, 2023-05-27:
I attached my model for you to try it yourself

`ton@debian:~/snap/freecad/common/Mod/Assembly4$ git branch --set-upstream-to=origin/bom-improvements-with-gui
Branch 'bom-improvements-with-gui' set up to track remote branch 'bom-improvements-with-gui' from 'origin'.
ton@debian:~/snap/freecad/common/Mod/Assembly4$ git pull
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Updating 7cb415e..f585218
Fast-forward
 Asm4_Measure.py                                  |   76 ++++----
 Asm4_libs.py                                     |  149 ++++++++++-----
 Asm4_objects.py                                  |  351 ++++++++++++++++++++++++++----------
 CHANGELOG.md                                     |   25 +++
 FastenersLib.py                                  |   52 +++++-
 HelpCmd.py                                       |   89 +--------
 INSTRUCTIONS.md                                  |  106 +++++------
 InfoKeys.py                                      |  234 ------------------------
 Init.py                                          |   30 ++--
 InitGui.py                                       |  178 +++++++++---------
 README.md                                        |   13 +-
 Resources/icons/Asm4_Export_PartsList.svg        |   63 +++++++
 Resources/icons/Asm4_Interference_Check.svg      |   61 +++++++
 Resources/icons/Asm4_Interference_Cleanup.svg    |   67 +++++++
 Resources/icons/Asm4_List_Liked_Files.svg        |   71 ++++++++
 Resources/icons/Asm4_List_Liked_Files_Tree.svg   |   78 ++++++++
 Resources/icons/Asm4_enableLinkSelection_b64.txt |    1 +
 Resources/icons/ExpressionEngine.svg             |  135 ++++++++++++++
 Resources/media/Patreon_Donate.svg               |  132 ++++++++++++++
 Resources/media/PayPal_Donate.svg                |  142 +++++++++++++++
 TECHMANUAL.md                                    |  132 --------------
 VERSION                                          |    2 +-
 VariablesLib.py                                  |   21 ++-
 checkInterference.py                             |  240 +++++++++++++++++++++++++
 checkInterference_lh.py                          |  240 +++++++++++++++++++++++++
 checkInterference_z.py                           |  194 ++++++++++++++++++++
 exportFiles.py                                   |  315 ++++++++++++++++++++++++++++++++
 gotoDocumentCmd.py                               |    5 +
 importDatumCmd.py                                |   33 ++--
 infoKeys.py                                      |   35 ++++
 infoPartCmd.py                                   |  361 +++++++++++++++++++++++++++++--------
 insertLinkCmd.py                                 |   51 ++----
 makeArrayCmd.py                                  |  183 ++++++++-----------
 makeBinderCmd.py                                 |   49 ++---
 makeBomCmd.py                                    | 1156 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-----------------------------------------
 mirrorPartCmd.py => mirrorPartCmd_obsolete.py    |    0
 newAssemblyCmd.py                                |   28 +--
 newDatumCmd.py                                   |   42 +++--
 newModelCmd.py                                   |  106 -----------
 newPartCmd.py                                    |   29 +--
 package.xml                                      |    6 +-
 placeDatumCmd.py                                 |  454 ----------------------------------------------
 placeLinkCmd.py                                  |    8 +-
 placeLinkUI.py                                   |   92 +++++-----
 placePartUI.py                                   |   51 ++++--
 releaseAttachmentCmd.py                          |   11 +-
 variantLinkCmd.py                                |   23 ++-
 47 files changed, 3769 insertions(+), 2151 deletions(-)
 delete mode 100644 InfoKeys.py
 create mode 100644 Resources/icons/Asm4_Export_PartsList.svg
 create mode 100644 Resources/icons/Asm4_Interference_Check.svg
[... 24 lines cut ...]

`[code]
OS: Ubuntu Core 20 (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33316 (Git) Snap 717
Build type: Release
Branch: master
Hash: a182c0140926e3682d1c305fceac632eee09d408
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * dodo 1.0.0
  * Render 2023.2.5
  * fasteners 0.4.56
  * FeedsAndSpeeds 0.5.0
  * Dracula 0.0.4
  * Defeaturing 1.2.0
  * POV-Ray-Rendering
  * Behave-Dark-Colors 0.1.1
  * FreeCAD_Assembly4 0.12.5
[/code]

[benb.FCStd.txt](https://github.com/Zolko-123/FreeCAD_Assembly4/files/11581245/benb.FCStd.txt)

`

leoheck, 2023-05-27:
Thank you, for having the time to check my fork and also for sharing your model. 

> The autofill does not work, maybe because of my model?
The Generate part list and Create Bom do nothing

I should look at the Parts info closely. I did not pay attention to it, yet. I was just trying to walk through the objects hierarchy and got give a bit of control to the user. Right now, it should be getting info from the Objects tree directly. But this is definitely something that I have to check. 

====

Ok, talking about the issue, this was just a variable with the wrong name that I could not catch with my sample tests. I've pushed a fix and this is what I am seeing now. 

[image]

Do you confirm it is what you are expecting to see?

snelweg, 2023-05-27:
Great, will check when I am home. Thanks!

snelweg, 2023-05-27:
>Do you confirm it is what you are expecting to see?

Yes, this is going the right direction. I still do not understand why the autofill of the fields does not work. Would be nice if it did.

The BOM is complete and even the Qty is correct. Very nice!
Also nice to be able to see what is going on if I switch on the 'verbose' checkmark.

Nice progress!

leoheck, 2023-05-27:
It's small step for the humankind, but a big step for the mechanical engineer haha

What the autofill should do? Put the values in the Part Info, so you can check and edit them yourself? I never needed that thing but if this is correct I will try to make it work.

snelweg, 2023-05-27:
> It's small step for the humankind, but a big step for the mechanical engineer haha

You are doing a great job. A working BOM is something that will make Asm4 a true assembly wb.

> What the autofill should do? Put the values in the Part Info, so you can check and edit them yourself? I never needed that thing but if this is correct I will try to make it work.

I understand that the autofill button will give default values retrieved from the  model. The length of a sketch and the width of the pad command I believe. Would be nice to have that working and not to have to fill out the values yourself. Would be supergreat if these values are updated if the model is changed.

https://github.com/Zolko-123/FreeCAD_Assembly4/tree/master/Examples/ConfigBOM

There are some real goodies in this fork: https://github.com/Ediforce44/FreeCAD-Assembly4-BOM The cutlist functionality is great for woodworkers like myself. Maybe you can cooperate with Ediforce44?
There on his git is a nice explanation of how models should be composed to automate the BOM and cutlist functions. For example:

In order to work properly for the new BOM you should follow the following tips:

    Keep everything clean:

        Every Part should contain a Body
        Every Body should contain a Pad with a Sketch inside

        The reason for that is that the BOM calculates the dimensions based on a virtual box arround the Pad of an Body.

    Keep your parts flat:

        Every Part should be flat, like if you saw it out of a wide plank

        Otherwise the dimensions and the assignment to a Cut-List get fucked up.

    Use your pad for thickness:

        Always draw your Part from the top. This means your Sketch has to be the top down view of the Part in your Cut-List.

        In other words:
        The length of the Pad of your Sketch has to be the thickness of the Part, otherwise the BOM doesn't assign the Part to the correct Cut-List.

        A loosely alternative is to draw your Sketch on the XZ-Plane or YZ-Plane and use the Z-Axis as thickness of your Part.

Most of these tips are for woodworkers, but will translate to other disciplines  I guess.

leoheck, 2023-05-27:
Those values are being added to the bom. I just don't know if they are right since they are meaningless for me. 

The infoPart menu, is not filling this info to the part, I guess. But the create bom, command is adding them to the spreadsheet. 

I could not fix this infoPart, thing yet since I don't understand how it should work or how it is used. 

I am not mechanical engineer, but computer engineer, so for me, filling these things automatically are the thing I was looking for because I don't want to have to customize every single part on my designs. But once I can understand how and why this infoPart works, I am going to fix this, definitely.

leoheck, 2023-05-28:
Major Ton, (@snelweg). I made some progress on the Auto Fill.

As you can see, I fill the Info using the `Autofill` button, then I edited the `Part_Label` 

[image]

Here the generated BOM
[image]

Please, update your copy of my fork, and let me know if it is a good start.

snelweg, 2023-05-29:
The autofill function works now, strange values though. 

~~But the BOM function does no longer work. I attached my model as it is right now, maybe I made a mistake.~~

I did make a mistake, only the 'active' parts in the assembly are put in the BOM. I did not expect that. Would be nicer if the type field of the part 'decides' if the part is included in the part list. Maybe some kind of hyrachie as in other BOM solutions? That way different BOM's for different type of parts (i.e. screws and fastners and another BOM for the materials).
That can also be done in a spreadsheet I know. But doing it straight from the source, so in FC, would make the chance of errors much more unlikly.

Great to see how fast there is progress. Hopefully @Zolko-123 will regonize the work and let it merge into the Main branch so everybody can enjoy and use your hard work.

Zolko-123, 2023-05-29:
> Great to see how fast there is progress. Hopefully @Zolko-123 will recognize the work and let it merge into the Main branch so everybody can enjoy and use your hard work.

yes what you're doing is great and of course it will be merged into main. BUT, because there is always a BUT:

Because I don't use BOMs very much, could you please provide some sort of documentation for this function ? What is the correct workflow ? What are the keywords ? 

From what I know, I'd suggest the following fields:

- Name
- Type
- Description
- Reference
- Revision
- Project
- Reference 2 (optional, for supplier reference for example))
- CutLength = max (BoundingBox[X], BoBoundingBox[Y], undingBox[Z] )

See a screenshot of the PartInfo in SolidWorks :

[image]

What do you think ?

snelweg, 2023-05-29:
> Because I don't use BOMs very much, could you please provide some sort of documentation for this function ? What is the correct workflow ? What are the keywords ?

Well for starters there are different needs for different disciplines. I need a functioning BOM for my woodworking, I know that metal workers might have different needs.
In the days when I used Fusion360 I was very happy with what OpenBOM did for me. But now they followed the AutoDesk route and want money for stuff that was always free.

As a woodworker an export to nesting/cutlist software would be very nice. Configurable fields would be necessary to export in one-go.

I used these cutlist solutions with Fusion back in the dark times:
https://sourceforge.net/projects/ctmc/
https://github.com/mru00/cutlet
https://www.optimalon.com/cutting_optimization_library.htm
Nowadays there are  many on-line solutions for optimized cutlist (nesting).

> From what I know, I'd suggest the following fields for PartInfo :
> 
>     * Name
> 
>     * Author

These can be extracted from the preferences in FC. I do not see any use for this information. A Bill of Materials does not need this I think, but it could be handy.
 
>     * Type

This would be a most useful field. That way separate lists can be made for wood, fasteners and such.

>     * Description

This could be a field to describe the function of the part

I would suggest a 
* Name field
This could be extracted from the body or part name (because we all give meaningful names to our bodies...)

>     * Reference

Like a project number?
 
>     * Revision

Sounds nice but I do not know how this would work in practice. Will this field be updated when a part is changed?

>     * Project
 
I do not understand the workflow here. Seems to me that the project is the title of the spreadsheet BOM. 

>     * Reference 2 (optional, for supplier reference for example))

That is what OpenBOM and all the others are for. You just connect your part name one time to all the supplier data. Next time the link will already be there. The data (price and such) are updated at a central point.
For a stand alone solution (as a woodworker) I do not see the need for this data.
 
>     * CutLength = max (BoundingBox[X], BoundingBox[Y], BoundingBox[Z] )

Proper values for the length and width field would be nice too. Auto updating when a part or body is altered. 
> 
> See a screenshot of the PartInfo in SolidWorks :
> [image]

I worked with Solidworks and was not very impressed by their BOM functions. I stopped working with it.
Sketchup worked better then SW :)

> What do you think ?

- The  material properties as used in the Arch WB could be usefull too here.  https://wiki.freecad.org/Arch_SetMaterial
- Maybe some template function so that we can export exactly what other programs expect.

I think that it is great that this all is happening so quickly, many thanks to @leoheck and @Zolko-123 of course! 
We are on the right track in my opinion.

leoheck, 2023-05-29:
I don't use PartInfo. Having to edit parts is to much for me. I just wanted a list of parts, quantities and Fasteners. But I am trying to make a progress here.

Zolko, on my fork I reverted the fields names back to the original since I was not used to the new names, and I had a lot of issues spread all over the place. So to make things easier I put the original fields names back. But I don't like them either. I will revert or improve this once I have a better BOM being generated.

Next thing I will check is the recording of arrays. I didn't check if this is working fine yet. 

If you have samples of files with features that results in a poor BOM list, please, share them with me here so I can check and make progress on these issues faster.

In short, for my application I just need a list of parts. 
I don't even need the Lengths, height or volume. I do reuse a lot of parts from the internet to make products. I don't want to make strict rules about the parts/bodies. But the user could tweak this in the GUI, I guess.

For instance, I added the Type_Label  there, so I can manually tell if the part is a Fastener even if it is a Part or Body and not a real Fastener from the Fasteners workbench.

snelweg, 2023-05-29:
> I don't use PartInfo. Having to edit parts is to much for me. I just wanted a list of parts, quantities and Fasteners. But I am trying to make a progress here.

What you have working now is very promising. I use partinfo for specific usages, for example the type of wood or maybe a certain finish. Partinfo is handy for that. Like the Material info found in Arch WB.

> Next thing I will check is the recording of arrays. I didn't check if this is working fine yet.

That doesn't work in this fork https://github.com/Ediforce44/FreeCAD-Assembly4-BOM
Will check if it works in yours.

> If you have samples of files with features that results in a poor BOM list, please, share them with me here so I can check and make progress on these issues faster.

My current project is a nice test. I will add fasteners to make things complicated. When I have something to show I will share.
 
> In short, for my application I just need a list of parts. I don't even need the Lengths, height or volume. I do reuse a lot of parts from the internet to make products. I don't want to make strict rules about the parts/bodies. But the user could tweak this in the GUI, I guess.

I think with a little more effort we could make the BOM functionality everybody is waiting for :)

> For instance, I added the Type_Label there, so I can manually tell if the part is a Fastener even if it is a Part or Body and not a real Fastener from the Fasteners workbench.

That's a good thing and it works.

leoheck, 2023-05-29:
Indeed. I like you shared your sample projects since I have just mine and then it would be biased for my needs indeed.

Zolko-123, 2023-05-29:
i can't check but I think it's obsolete and forgotten

closed 2023-05-29 by Zolko-123