Zolko-123/FreeCAD_Assembly4 #399: Fasteners seem not to be working anymore.

leoheck, 2023-02-12:
I select a circular profile, then I use "New Hole Axis", then I use "Insert Screw" and the screw is not placed in the right place.

leoheck, 2023-02-12:
This is from dev branch, 

```
17:38:09  Initializing Assembly4 workbench (v0.12.5) ............................ done.
17:41:29  Running the Python command 'Asm4_newHole' failed:
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/FreeCAD_Assembly4/./newDatumCmd.py", line 215, in Activated
    parentDoc  = parentPart.Document

'NoneType' object has no attribute 'Document'Running the Python command 'Asm4_newHole' failed:
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/FreeCAD_Assembly4/./newDatumCmd.py", line 215, in Activated
    parentDoc  = parentPart.Document

'NoneType' object has no attribute 'Document'Running the Python command 'Asm4_newHole' failed:
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/FreeCAD_Assembly4/./newDatumCmd.py", line 215, in Activated
    parentDoc  = parentPart.Document

'NoneType' object has no attribute 'Document'Running the Python command 'Asm4_newHole' failed:
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/FreeCAD_Assembly4/./newDatumCmd.py", line 215, in Activated
    parentDoc  = parentPart.Document

'NoneType' object has no attribute 'Document'
```

Zolko-123, 2023-02-13:
yes something has changed there, I have noticed too. What are the exact commands you have made ?

Also, what should be the expected behaviour : when a hole-axis is selected, create the fastener in the same part container and on that hole-axis ?

Zolko-123, 2023-02-13:
the error from new axis-hole comes because you have tried to do that on an object outside of a container. This error is intercepted now, thanx for the report

The fasteners placement is not done, but this is another issue. In the future, can you please avoid putting 2 issues in 1 report please ?

leoheck, 2023-02-13:
I tried a bunch of things, on 2 versions of FreeCad (latest snap and AppImage), I also reinstalled Asm4 to the latest to make sure I am not doing anything wrong, and I also changed to the development branch. Most of the time I am very scared to even ask things here since most of the time my issues are not welcomed without spending weeks on Freecad Forum. But the Freecad forum blocked me a couple of days ago, and I cannot log in there anymore. That forum is horrible, by the way.  

What I tried in using Freecad was, selecting the circular profile inside the Part Design Wb, with the Body Active, and creating the Circular Axis, some times a good object is created, some times it is broken (with a broken symbol) for an unknown reason. I also tried to import the Circular Axis to the main Model, but it did not work either. I also tried to use Fasteners from Asm4 Wb and also from Fasteners since Asm4 is missing many things and also American screws. 

But since we are talking again, I am going to explore this issue better to report it deeply. Being able to use screws, easily, is a really important feature.

Zolko-123, 2023-02-13:
> What I tried in using Freecad was, selecting the circular profile inside the Part Design Wb, with the Body Active, and creating the Circular Axis

Does this have anything to do with Assembly4 ?

leoheck, 2023-02-14:
yes it does.

Zolko-123, 2023-02-20:
So ... what was working before and doesn't work anymore ? Can you please be specific ?

leoheck, 2023-02-20:
**TLDR**; there are 2 issues. One is my fault, having the fastener being created outside of the Model and not being able to identify this visually since the default Classic theme is bad as fuck. And The second is the Freecad issue that is changing the Display Mode Body to Tip, automatically for no reason. Then, because of that LCS's and other coordinates are constantly not being displayed while positioning the Fastener. 

Because of that, none of those things are related to Asm4. I am closing this. 

----

I am selecting the hole profiles, with the part in the Parts folder.
Then I am using the "New Hole Axis" from Asm4 toolbar.

[image]

Then, I can immediately see those hole axis, inside the link of my Part, inside the Model.

[image]

Then I create the fastener.

**My first problem** is that it was being placed outside of the model. 
And this is barely perceptive since it is not easy to see they are nested inside the Model. This is the user issue, not of Asm4.

Now, this dark theme that I am using since yesterday, allows me to visualize this easier. 

Fastener outside of the Model
[image]

Fasterner inside of the model
[image]

Using this dark theme... 

Fastener outside of the Model
[image]

Fasterner inside of the model
[image]

**The second problem** I am having is that Freecad is changing the Display Mode Body of each Part to `Tip` **automatically after editing or doing something with that Part**. I did not know this was happening and this is annoying as fuck. It may be a bug. I never ever use this `Tip` mode. 

With that being said, it looks like Asm4 is not the culprit of my issues. Thanks for assisting me to discover this Display Mode Body issue. I am closing this issue now.

Registering Freecad versions I have installed here:

```
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31964 (Git) Snap 617
Build type: Release
Branch: master
Hash: cd88d3888c3025424332b674940c07ac2978a1ea
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.54
  * Assembly4 0.12.5
  * Part-o-magic 1.0.0
  * freecad.gears 1.0.0
  * ProDarkThemePreferencePack 1.0.0
  * kicadStepUpMod 10.17.1
```

```
OS: Ubuntu 22.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git) AppImage
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.6
  * fasteners 0.4.54
  * lattice2 1.0.0
  * FreeCAD_Assembly4 0.12.5
  * Assembly4 0.12.5
  * kicadStepUpMod 10.17.1
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
```

closed 2023-02-20 by leoheck

leoheck, 2023-02-20:
Another issue of mine is that the fucking forum is blocking my user account. I created a second one and it was blocked too. I also cannot send emails to the Admins through the Forum either to be able to discuss or report these things. =/