Zolko-123/FreeCAD_Assembly4 #400: Trying to follow FreeCAD_Assembly4/INSTRUCTIONS.md, "Workflow" section, no macro "new_Model"

ssteinerx, 2023-02-14:
[code]
OS: Ubuntu 22.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Python 3.10.7, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * Assembly4 0.12.5
  * Dracula 0.0.2
[/code]

Installed through Addon Manager.

I've never used a macro, but there aren't any listed in Macro -> Macros in either User macros or System macros.  User macros location is: ~/.local/share/FreeCAD/Macro

At the very end in Tutorials and Examples, the tutorials show the use of the Assembly menu so I'm guessing the macro instructions are just out of date/left over from Assembly[1-3] or something?

Zolko-123, 2023-02-14:
where did you read that it's a macro ?

ssteinerx, 2023-02-15:
https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/INSTRUCTIONS.md#part

Part

The basic workflow for creating a part is the following:

    Create a new document, create a new 'Model' with the **macro** 'new_Model'. This will create a new App::Part called 'Model' with the default structure

Zolko-123, 2023-02-15:
ah yes, that's some old text. It's a command now, either in the menu or the toolbar

closed 2023-04-11 by Zolko-123