Zolko-123/FreeCAD_Assembly4 #401: dynamic reference to varTmpDoc

juddidier, 2023-02-17:
Hi,
I try to implement the variant part into ASM4. I've tweaked the code to almost success till now. The only problem that I still have is when having to handle the dynamically created varTmpDoc from the variant part.
To show the problem I made a simple example with 2 custom ball bearings, where the second is attached to the first over LCS:
[image]

The problem arises in the "Placement" property of `bearing_2` when reloading (after FreeCad restart) the assembly. The expression reference to two `varTmpDoc`, one to itself and one to the linked object. Since the document number is dynamically assigned, the number might change and the expression gets broken:
[image]

How would the path look like through the class hierarchy beginning from obj... (in the example `bearing_1`), in the way that there is no more "varTmpDoc_xxx' in the expression?
Is this possible at all?

[testassembly.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/10766056/testassembly.zip)

Zolko-123, 2023-02-17:
no, sorry, this is an inherent limitation of variant link. You have to think differently.

Since the dimensions of the variant link part are given in the assembly by the very definition of a variant link, use that information to place an LCS in the assembly where the corresponding LCS would be in the variant link. Think of it, it works.

What you want to do is both complicated and error prone.

juddidier, 2023-02-17:
> no, sorry, this is an inherent limitation of variant link. You have to think differently.

Hmmm, I like to think "what belongs together, should be together"... 

 
> Since the dimensions of the variant link part are given in the assembly by the very definition of a variant link, use that information to place an LCS in the assembly where the corresponding LCS would be in the variant link. Think of it, it works.

You mean, depending on what is (in my example) in `VariantVariables/bearingType` I should calculate the position of an external LCS and attach my Variant link to this LCS?
But this solves the problem only by half, since the 'place link' function attach LCS to LCS, I still need at least one LCS in the variant link and the varTmpDoc to itself stays in the Expression engine.

Zolko-123, 2023-02-17:
The link knows its _LinkedObject_ and thus doesn't need any reference to which _varTmpDoc_ that _LinkedObject_ is. So yes, you have to figure out from the assembly where the other LCS in the variant link is.

Honestly, having a catalog of bearings wasn't the first motivation to do the variant link, so you're exploring new territory. Think swamps and dragons, may-be you'll be lucky to encounter a princess. Or may-be not

juddidier, 2023-02-17:
> The link knows its _LinkedObject_ and thus doesn't need any reference to which _varTmpDoc_ that _LinkedObject_ is. So yes, you have to figure out from the assembly where the other LCS in the variant link is.

Well, unfortunately in the way I use them for the parametric parts it is not absolutely true. The enum list gets its items from the spreadsheet with a
`varTmpDoc_1#<<Parameter>>.cells[<<A3:|>>]`
first I added a function in `onDocumentRestored` of ASM4_objects.py that finds `varTmpDoc` in the expression engine and updates the link. This worked until I added the LCS.

I tried to add the parametric parts into the assembly in quite a few different ways, but none worked except adding the parametric parts as variant link into the assembly. My goal is to use the assembly with the `exploded assembly` and `techdraw` workbenches and here the stuff gets tricky. Since my final assembly will have more than 100 parts, I need something more efficient than doing all "by hand".

juddidier, 2023-02-20:
Hi Zolko,

I had two night session with trial-and-error and came out with a solution that seems to work (mostly). The changed expression from the example above looks like:
[image]
--> take care of the 'dot' after LCS-Name!

I'm not 100% happy, since I had to remove the part name in favour of the label (which must not be unique), and the label must have the same restrictions in allowed characters as the name "_", "0..9", "a..z", "A..Z"
Actually when I reopen "Edit placement of part" the predefined chose is lost on the variant link part, since the expression doesn't contain the name anymore. Will have also to tweak the "_init_" function.

I saved my actual work under [https://github.com/juddidier/Assembly4](url)

Zolko-123, 2023-04-15:
OK, so I close it for now

closed 2023-04-15 by Zolko-123