Zolko-123/FreeCAD_Assembly4 #402: XYZ translation is not accepting floating numbers and formula

leoheck, 2023-02-18:
I cannot put floating numbers. 
I am trying with . (US format) and with (Brazilian format)
None of these work. 

The dot is added, but it is not possible to add values after that.

This field does not accept equations using the "=" shortcut.

[image]

I imagine that Freecad has a special input field for this kind of thing that could be used here, easily.

Zolko-123, 2023-02-20:
it's probably a setting in your locale, `.` –vs– `,` , on my computer it works

> I imagine that Freecad has a special input field for this kind of thing that could be used here

yes, I think

> easily.

If it's "_easy_" for you thank-you for submitting a PR

leoheck, 2023-02-20:
Yesterday I installed a Theme for the first time. Then this started to work again. 

DarkPro
[image]

Classic theme
[image]

Freecad behaves so badly sometimes.

closed 2023-02-20 by leoheck