Zolko-123/FreeCAD_Assembly4 #411: Version 0.12.6 won't install

neildarlow, 2023-02-27:
The Addon manager reports a failed update for Assembly4-0.12.6. Previous versions haven't been a problem.

````
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.12.5
  * Curves 0.6.8
  * fasteners 0.4.54
  * lattice2 1.0.0
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
````

I'm not sure what the Addon manager is telling me here. It says there's an update available for Assembly4-0.12.6 but it also says that version is installed. Is the metadata broken?

[image]

Zolko-123, 2023-02-27:
I just checked on my computer (I removed the Assembly4 directory by hand) and it installs without problems. Can you remove your v0.12.5 and install again ?

```
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.30890 (Git)
Build type: Compiled
Branch: master
Hash: 1eaeee5f7a63472de04c5601c7ffb924473b8de1
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.56a
  * CurvedShapes 1.0.3
  * fcgear 1.0.0
  * sheetmetal 0.2.50
  * TabBar
  * Render 2022.2.0
  * Curves 0.5.1
  * fasteners 0.4.4
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
```

Zolko-123, 2023-02-27:
But your right, the info reported by the _Addon Manager_ is wrong:

[image]

Try updating the local cache of the Addon Manager

neildarlow, 2023-02-27:
I had refreshed the Addon manager local cache. I was able to Uninstall/Restart FreeCAD/Install Assembly4 using the Addon manager.

So, it looks like there was an issue with updating for this particular version of Assembly4 but fresh installs are OK. I guess it could be due to a bug in Addon manager but it's the first time I've encountered this.

Thanks for responding.

closed 2023-02-27 by Zolko-123